Arksine/moonraker #100: Thumbnail preview from Ideamaker Slicer

modmt, 2021-02-23:
Hello, 
I want go ask if it is possible to extract a thumbnail preview from an Ideamaker sliced file to show it in Mainsail?
I asked this also in the Mainsail Git.
https://github.com/meteyou/mainsail/issues/180

If i understood right there is the possibility in Ideamaker to wrote Out an additional .Data File, which contains a Picture for displaying the model on the Raise3D Machines. But I do not know how this could be used.

Could you help me with this?
I can provide you an Ideamaker files if necessary.

Thank you in advance.
Matthias

Arksine, 2021-02-26:
Unfortunately there is little I can do with this without some sort of specification.   The data file is a binary.  There is some Base64 encoded data at the end of the gcode file as well.  Without some guidance as to how to interpret this data it isn't possible to do anything with it.

modmt, 2021-02-26:
OK, thank you. No problem. Maybe I have to ask the Ideamaker team how to handle the file (or at least if it's possible to extract an picture). I will come back to you if I know more.

Arksine, 2021-05-10:
Since discussion here seems to have concluded I'm going to go ahead and close this issue.  If ideamaker wants to provide details on how to parse their thumbnails I'm open to adding support in Moonraker.

closed 2021-05-10 by Arksine