Arksine/moonraker #101: power plugin ignores initial_state option in sections other than gpio type

photocromax, 2021-02-26:
I just installed moonraker (in fluidpi) a few days ago and I noticed "initial_state" option is not handled in tasmota/tplink/shelly power type devices and, of course, I would like to use a tasmota device :-)

I tried to fix it myself adding the option configuration in  tasmota __init__ and initialize methods, but I cant find where the valid options are specified. "initial_value" is stripped off in the parsed configuration "/server/config"

Since config validation was added I get a warning message in moonraker log saying "initial_value" is not a valid option for my power/tasmota section

Can someone point me out in the right direction? Or can this feature to be added?

Thank you in advance
PhotoCroMax

Arksine, 2021-02-26:
The initial state option is not valid for non-gpio devices.  These devices are able to retain their state and report it upon initialization, thus it is not required to initalize their state.  GPIO devices cannot do this, the GPIO lines are released when moonraker shuts down.

photocromax, 2021-02-26:
Thans for your reply Arksine.

I thought initial_value was intended to force a status at boot, not just for retains reasons.

Maybe it could be useful to have something like "force_value" (with "none" as a default), and power plugin could initialize the status only if "force_value" is on or off.

Restarting Moonraker will automatically configure a clean state, ignoring the retained status from previous situations.

If you think this could be a safety issue, please disregard this idea.

Thank you
PhotoCroMax

Arksine, 2021-05-10:
I'm going to go ahead and close this as I don't believe setting an initial state for non-gpio devices would be the appropriate thing to do since stopping Moonraker has no affect on their state.

closed 2021-05-10 by Arksine