Arksine/moonraker #103: Merge with timelapse plugin fork

cinadr, 2021-03-02:
Hi!

Is there any chance to merge [this](https://github.com/FrYakaTKoP/moonraker) into master? This fork contains [timelapse ](https://github.com/FrYakaTKoP/moonraker/blob/dev-timelapse/docs/configuration.md#timelapse-plugin) plugin which is a major feature.

Thanks.

Arksine, 2021-03-06:
I'm going to go ahead and close this, as we need to wait for the timelapse dev to decide when he wants to merge.

closed 2021-03-06 by Arksine

FrYakaTKoP, 2021-03-08:
@cinadr the plugin works, but without frontend implementation it's not pleasant to use (need to download the videos via external app etc.) 

The PR is not far away, i just like to revisit the code and clean it up a notch and complete the docs. 

best timeframe i can give is soon™

-FrY

cinadr, 2021-03-08:
Thank you for your update. I'm patiently waiting to test this. Hopefully web frontend developers come up with some  download solution to take advantage of this plugin soon.

FrYakaTKoP, 2021-03-08:
@cinadr you actually can test already test it

- either you clone my fork or just copy the timelapse.py (be warned you may loose update manager functionality by doing so but going back is no problem)
- and then follow this doc: https://github.com/FrYakaTKoP/moonraker/blob/dev-timelapse/docs/configuration.md#timelapse-plugin

Let me know if you got problems. feedback is also very much appreciated.

Don't worry atleast one frontend dev is onto it ;)

ayufan, 2021-03-08:
@FrYakaTKoP 

I think the lack of frontend is usually not a problem as a first iteration :) Someone would contribute frontend next!

meteyou, 2021-03-08:
@ayufan 
i am currently testing if everything is supported for a web interface. the first implementation worked but was not usable for the user via the webinterface.

cinadr, 2021-03-10:
Hi!

I managed to use this without braking the update funcion. And it is working well so far. An option to specify folder paths would be nice as my /tmp folder is right now resides on a RAM disk for sparing my sd-card and it fills quickly.
Thank you.

FrYakaTKoP, 2021-03-10:
@cinadr 
interesting i was used to believe /tmp lives in tmpfs (RAM) anyways, strange that raspbian (rpi OS) does different, may because of the little RAM on some rpis.

i just pushed configurable snapshot directory ;)
https://github.com/FrYakaTKoP/moonraker/commit/6eed16733f5995c5a915c55c21c4c60343a0bffb

moebis, 2021-05-26:
@FrYakaTKoP is there an ETA for when time-lapse feature will be available. I'm almost considering going back to Octoprint because I miss it so much. Well, it is only a sd swap away, I suppose I can just swap cards when I want to do a time-lapse.

FrYakaTKoP, 2021-05-26:
there is no ETA when it get merged to the main branch, but you can use it as is. Just copy the python script to your moonraker install and set it up like documented and it should work (didn't tested with the last couple commits, but alot of other people use it this way and i didn't hear from anyone it breaked, if so i will try to fixit). Also you can exclude the timelapse.py from the update manager to preserve update functionality of the main branch (obviously excluding updates to timelapse.py). 

The merge will hapen after the frontend guys had time to implement it properly, but they have other (far more important) priorities. Also i'm short on time right now too, so i can't keep my fork uptodate (or implement more features to the plugin).

matthewlloyd, 2021-06-12:
Can you please merge this without frontend support? It's still very useful even without a frontend. A lot of people would find this useful, but don't know about this issue and would never stumble upon it. It's so easy to configure, it takes literally 30 seconds to set up. It's trivial to download and view the videos with an SCP client (e.g. WinSCP). A frontend would be nice but it's not essential.

ayufan, 2021-06-12:
I posted this design change to unify camera support and describe a way to integrate post-print snapshot and timelapse: https://github.com/Arksine/moonraker/issues/178.

schnibbi, 2021-08-29:
I tried to go with manual installation as mentioned in the docs. Unfortunately i get a moonraker error saysing there is a error in the timeplase section. Maybe because Mainsail and Moonraker just switched to V 2.0.0 ?

FrYakaTKoP, 2021-08-29:
@schnibbi

i'm using uptodate moonraker and mainsail without a problem. Did you copy/install the timelapse.py?

just to make sure, connect to ssh and do following:
```
cd ~/moonraker/moonraker/components/
curl -O https://raw.githubusercontent.com/FrYakaTKoP/moonraker/dev-timelapse/moonraker/components/timelapse.py
sudo systemctl restart moonraker
```
That will download/update the script from my fork and restart moonraker. After that refresh your browser and check if there is still a error. If so please upload the your moonraker.log so i can analyze it.

schnibbi, 2021-08-30:
Thanks i found that commands on discord top, you already discussed with another User. Strange Thing is it did wörk the first time than no more. I then Set everything up from scratch an now its working just fine.Thanks for that awesome Feature! You also mentioned you are open for other Features. Im Not quiet Sure if its possible but IT would be Nice to have more picture per Layer maybe like a percentual amount of gcode per layer so you can havo longer videos Von meinem/meiner Galaxy gesendet

FrYakaTKoP, 2021-08-30:
@schnibbi  
you would need to add the macro into your slicer in mid of layer where you want a snapshot to be taken it's not in my hand. But you could add timebased frame grabbing described here:
https://github.com/FrYakaTKoP/moonraker/blob/dev-timelapse/docs/configuration.md#timebased-frame-capture

this would capture a frame every x seconds

aeremichev, 2021-09-01:
> @schnibbi
> you would need to add the macro into your slicer in mid of a layer where you want a snapshot to be taken, what i don't think is possible with most slicers. But you could add timebased frame grabbing described here:
> https://github.com/FrYakaTKoP/moonraker/blob/dev-timelapse/docs/configuration.md#timebased-frame-capture
> 
> this would capture a frame every x seconds

Hey!
Thanks for your work.
I have a question.
Does moonraker have a layer change event?
Is it possible to make a snapshot of this event?
How it is implemented in OctoPrint.

FrYakaTKoP, 2021-09-01:
@aeremichev 
no Moonraker doesn't have such a event. Moonraker doesn't read the Gcode. Klipper does read the gcode directly, but detecting a "layer change" isn't easy, since change in Z could also be a "ZHop".

Octoprint does this with some trickery, it needles out zhops. I had used it as back when i was using octoprint and it works most of the time.  That's no question but it's also not perfect. I had couple scenarios where it would not take a frame even it was a legit layer change.

To pull someting similar off, i would need to also add code to Klipper (afaik there is no such "layer change event" in klipper yet).  I am more then happy to add, documentation how to configure, if someone add such a event to klipper. I would not force people to add something to there slicers, if i'm aware of such feature 😉. To be honest: i was lazy and said myself why try to do something that you know isn't a bulletproof solution, just to save people the "hassle" to add the Macro to there slicers. Also it would be not part of the plugin itself, i could look into it later if heavily requested.

TL;DR
maybe there will something similar in future, but chances are high that it will never appear or will not be bulletproof (therefore not recommended)