Arksine/moonraker #106: TpLink Power strip does not work

Kaikhong, 2021-03-03:
I have a KASA power strip (former known as TP Link) and it does not work. The power plugin in Fluiddpi just report LED[error]. Does it mean that the power strip is not supported by moonraker? I thought the port in configuration means the port in power strip. Im pretty sure my IP address is correct, I could set up it with TPLink Smartplug plugin in Octoprint before.

Doridian, 2021-03-12:
`port` for that power device means the TCP, as in network, port. Not the port on the power socket.
So, no, the power socket is currently not supported.

Arksine, 2021-04-21:
This should have been resolved by #129.  Add the socket index to the ip address.  For example, to control socket 1:
```
[power]
type: tplink_smartplug
address: 192.168.1.123/1
```

closed 2021-04-21 by Arksine