Arksine/moonraker #111: Upload from PrusaSlicer not working anymore

nicedevil007, 2021-03-05:
Heya, I recently updated to the latest build of moonraker. Now PrusaSlicer isn't working anmore with direct upload / start print.

I get this message:
[image]

What I tryed to fix it:
Rebooting the printer (unplug/replug power)
Rebooting the raspberry pi (host reboot)
Restarting moonraker (service reboot)

Nothing helped so far. 

Uploading via Fluidd WebUI still works.

Arksine, 2021-03-05:
Hi.  You need add [octoprint_compat] to moonraker.conf and restart the service.  

For future reference it is a good idea to check [user_changes.md](https://github.com/Arksine/moonraker/blob/master/docs/user_changes.md) if you have an issue to make sure that isn't related to a recent config change.

nicedevil007, 2021-03-05:
> Hi. You need add [octoprint_compat] to moonraker.conf and restart the service.
> 
> For future reference it is a good idea to check [user_changes.md](https://github.com/Arksine/moonraker/blob/master/docs/user_changes.md) if you have an issue to make sure that isn't related to a recent config change.

sry, didn't saw this, added and working again :) Thank you!

closed 2021-03-05 by nicedevil007