Arksine/moonraker #112: OctoPrint support incompatible with SuperSlicer

cloakedcode, 2021-03-06:
I've enabled the new OctoPrint API support with `[octoprint_compat]` and now get this in SuperSlicer:

[image]

Fluidd lists Moonraker version `v0.3.1-42 `. I will likely downgrade while this issue is resolved to make my printer usable again.

Out of curiosity, are there Moonraker features that will make the slicer integration better than what the OctoPrint (compatible) API provides?

zellneralex, 2021-03-06:
See https://github.com/Arksine/moonraker/issues/111

Arksine, 2021-03-06:
What version of SuperSlicer is this?  I can confirm that uploads work with 2.2.53.  I can also confirm PrusaSlicer 2.3.0 works.

Please include moonraker.log.  My guess is that you added `[octorpint_compat]` without restarting the server.

Turge08, 2021-03-06:
I'm having the same issue with PrusaSlicer after updating Moonraker.

[image]

This was working earlier yesterday.

Arksine, 2021-03-06:
Yes, adding `[octoprint_compat]` and restarting the moonraker service should resolve your problems.  If you could report back that it is resolved would appreciate it.

Turge08, 2021-03-06:
> Yes, adding `[octoprint_compat]` and restarting the moonraker service should resolve your problems. If you could report back that it is resolved would appreciate it.

Yup it works after adding the entry to moonraker.conf and restarting moonraker. 

Thanks!

cloakedcode, 2021-03-06:
Very strange... I swear Fluidd restarted Moonraker when I edited the conf and hit the "Save & Restart" button. This time updating, adding `[octoprint_compat],` and hitting that same button resulted in SuperSlicer complaining with 403 Forbidden code. Oddly the log indicates the IP is detected as trusted but then later responds with 403 to that same IP. Restarting the service again (this time with the circular arrow restart button in Fluidd's sidebar) fixed the issue.

I've attached the logs. Should I open a separate issue for this? Is there a restart API endpoint that is possibly not doing a full tear down or something? It seems odd that I've run into several issues involving restarts that were fixed by subsequent restart.
[moonraker(3).log](https://github.com/Arksine/moonraker/files/6095972/moonraker.3.log)

Arksine, 2021-03-06:
You got a forbidden error because Klipper had the file loaded.  Moonraker will refuse to overwrite the file in this scenario.

Arksine, 2021-03-06:
Hmm, looking closer, Klippy doesn't appear to be connected.  Let me investigate this further.

cloakedcode, 2021-03-06:
Klipper loads all the *.conf files, not just printer.conf? Why is Klipper touching moonraker.conf?

Arksine, 2021-03-06:
You have it.  However I may modify the check in Moonraker to allow the file to be overwritten if Klippy is in the `shutdown` state.  Even though Klippy reports as having a the file loaded, it really doesn't when its shutdown.

Arksine, 2021-04-20:
I'm going to close this as it should be resolved.

closed 2021-04-20 by Arksine