Arksine/moonraker #113: Cura OctoPrint error

morganfw, 2021-03-06:
Hi all.
After installing new Moonraker version with initial support to send directly from Slicer (Cura with Octoprint plug-in in my case) to moonraker, added configuration

`[octoprint_compat]`

to moonraker.conf and restarting then configuring Cura OctoPrint plug-in, all works fine, the sliced project was sent to Moonracker and I see in the interface (Fluidd in my case) but Cura continually displays following error:

[image]

I do something wrong?
Thanks for the great work you are doing.

davide-molino, 2021-03-06:
same issue with superslicer

Marct27, 2021-03-06:
Seem to be having the same issue with Prusa Slicer
```
klipper
v0.9.1-295

moonraker
v0.3.1-42

mainsail
v1.3.0
```

Error from Prusa Slicer v 2.3.0
```
Error uploading to print host:
HTTP 404: Traceback (most recent call last):
File "/home/pi/moonraker-env/lib/python3.7/site-packages/tornado/web.py", line 1681, in _execute
result = self.prepare()
File "/home/pi/moonraker-env/lib/python3.7/site-packages/tornado/web.py", line 2430, in prepare
raise HTTPError(self._status_code)
tornado.web.HTTPError: HTTP 404: Not Found
```

Arksine, 2021-03-06:
@davide-molino @Marct27 You are not experiencing the same issue.  You need to add `[octoprint_compat]` to moonraker.conf and restart the service, as described in #111.  For future reference, please check `user_changes.md` if you have a problem.

The OPs issue is a bug in the module related to temperature reporting to Cura that will be fixed shortly.

Arksine, 2021-03-06:
@morganfw This should now be fixed in commit d108334

morganfw, 2021-03-06:
> @morganfw This should now be fixed in commit [d108334](https://github.com/Arksine/moonraker/commit/d108334c37f8ed7e7566e19e49b58503f289896a)

That sounds good, now I'm printing, when finished I will update moonraker and retry from Slicer.

Thanks.

morganfw, 2021-03-06:
I confirm, all works fine now.
Updated Moonraker, restarted service and now Cura with OctoPrint plug-in doesn't show more python traceback message.

Thanks for the quick fix.

closed 2021-03-06 by morganfw