Arksine/moonraker #114: Path error

SilverWx, 2021-03-06:
**ERROR :**
Unable to open config file /home/pi/klipper_config/home/pi/klipper_config/printer.cfg
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

**CONFIG : ** the default one 
moonraker.conf in /home/pi:/klipper_config
`[server]
host: 0.0.0.0
port: 7125
enable_debug_logging: True
config_path: /home/pi/klipper_config
klippy_uds_address: /tmp/klippy_uds

[authorization]
enabled: True
api_key_file: ~/.moonraker_api_key
trusted_clients:
    127.0.0.1
    192.168.0.0/16
    ::1/128
    FE80::/10
cors_domains:
    http://*.local
    http://my.mainsail.xyz
    https://my.mainsail.xyz
    http://app.fluidd.xyz
    https://app.fluidd.xyz
    http://192.168.1.26
    http://192.168.1.26:*

[octoprint_compat]

[update_manager]

[update_manager client mainsail]
type: web
repo: meteyou/mainsail
path: ~/mainsail

[update_manager client fluidd]
type: web
repo: cadriel/fluidd
path: ~/fluidd`

Arksine, 2021-03-06:
Please include moonraker.log

Arksine, 2021-03-06:
After a closer look, this doesn't appear to be an issue with Moonraker.  You seem to have a problem with your Klipper installation.  Check the installation documentation and make sure that your Klipper `defaults` file is set up correctly

SilverWx, 2021-03-06:
Okay, this happen after updating kiauh to v3 and upgrading moonraker, I will try to check what you said me.

SilverWx, 2021-03-06:
I finally reinstalled klipper and all are working now, Thank you for your help :)

closed 2021-03-06 by SilverWx