Arksine/moonraker #119: just updated to the latest moonraker via Fluidd and can no longer connect

tomohulk, 2021-03-12:
Just updated the moonraker repo via the Fluidd interface and now it will no longer connect.

is there any logs I can get over ssh to help out?

[image]

tomohulk, 2021-03-12:
this is my config moonraker config:

```
[server]
host: 0.0.0.0
port: 7125
enable_debug_logging: False
config_path: ~/klipper_config
temperature_store_size: 600
gcode_store_size: 1000

[authorization]
enabled: True
cors_domains:
  *.local
  *.lan
  *://app.fluidd.xyz
  *://192.168.*

trusted_clients:
  10.0.0.0/8
  127.0.0.0/8
  169.254.0.0/16
  172.16.0.0/12
  192.168.0.0/16
  FE80::/10
  ::1/128

[octoprint_compat]

[update_manager]
[update_manager client fluidd]
type: web
repo: cadriel/fluidd
path: ~/fluidd
```

my home network is a 10.0.0.0/24

Arksine, 2021-03-12:
Moonraker's log is available at `/tmp/moonraker.log`.  However this is likely due to a recent change that raises an error when an unsafe entry is detected in `cors_domains`.  SSH in to your pi, remove `*://192.168.*` from `cors_domains`, save the file, then `sudo service moonraker restart`.

For future reference, [user_changes.md](https://github.com/Arksine/moonraker/blob/master/docs/user_changes.md) attempts to document changes that result in a broken config.

tomohulk, 2021-03-12:
that was it.  I will make sure to do that from now on.  thanks for very quick response!

closed 2021-03-12 by tomohulk

ilyakyznesov, 2021-05-18:
uh, it worked ... I don’t know why it occurred to me, but I disconnected the arduino cable from the rasberry, waited until everything started, plugged it back in and it ... it worked! very strange, thank you all for your help, I hope this "method" will help someone