Arksine/moonraker #120: main is replacing master branch

gcormier, 2021-03-15:
https://github.com/Arksine/moonraker/blob/3b99307109d293f264b0abb1aa822d407f8367eb/moonraker/plugins/update_manager.py#L924

new github respositories use "main" as the default branch.

https://github.blog/changelog/2020-10-01-the-default-branch-for-newly-created-repositories-is-now-main/

update_manager will fail for repositories that choose to use this new convention (eg. https://github.com/eliteSchwein/mooncord )

gcormier, 2021-03-15:
```
2021-03-15 17:13:49,632 [update_manager.py:log_repo_info()] - Git Repo MoonCord Detected:
Owner: eliteSchwein
Path: /home/pi/mooncord
Remote: origin
Branch: main
Remote URL: https://github.com/eliteSchwein/mooncord.git
Current Commit SHA: 2dbd22f11d3bd39a7be574c880e105964b13481f
Upstream Commit SHA: 2dbd22f11d3bd39a7be574c880e105964b13481f
Current Version: ?
Upstream Version: ?
Is Dirty: True
Is Detached: False
Commits Behind: 0
2021-03-15 17:13:49,632 [update_manager.py:_log_info()] - Repo MoonCord: Repo validation checks failed:
Repo not on default remote branch: origin/main
2021-03-15 17:13:49,632 [update_manager.py:_log_info()] - Repo MoonCord: Updates on repo disabled
```

Arksine, 2021-03-15:
Thanks.  I will add a `primary_branch` option to the repo config that defaults to `master`.  That way it wont break existing configurations but allow for clients to use `main` or any other branch name as the primary.

Arksine, 2021-03-16:
This should now be fixed with commit 4b89de53f74a95181411c312cf88d5256ada4d30.

gcormier, 2021-03-16:
Awesome, I can test it when the next release goes out.

gcormier, 2021-03-17:
Working for me!

closed 2021-03-17 by gcormier

eliteSchwein, 2021-03-27:
hey there dev from mooncord, it will take some time till the next release :(