Arksine/moonraker #121: Corrupt git causing update manager issues

boardthatpowder, 2021-03-17:
Periodically update manager fails. Inspecting the logs, it is failing due too corrupt git. Update manager is the only process I'm aware of that is making changes via git to these projects.

Moonraker log extract:

```
--------------------Moonraker Log Start--------------------
Git Version: v0.4.1-33-g978fcff

#################### Moonraker Configuration ####################

[server]
host = 0.0.0.0
port = 7125
enable_debug_logging = True
config_path = ~/klipper_config
temperature_store_size = 600
gcode_store_size = 1000

[authorization]
enabled = True
cors_domains =
	*.local
trusted_clients =
	10.0.0.0/8
	127.0.0.0/8
	169.254.0.0/16
	172.16.0.0/12
	192.168.0.0/16
	FE80::/10
	::1/128

[update_manager]

[update_manager client mainsail]
type = web
repo = meteyou/mainsail
path = ~/mainsail

[octoprint_compat]

[system_args]
configfile = /home/pi/klipper_config/moonraker.conf
logfile = /tmp/moonraker.log
software_version = v0.4.1-33-g978fcff

#################################################################
2021-03-16 14:05:33,005 [update_manager.py:github_api_request()] - GitHub API Request Processed
URL: https://api.github.com/rate_limit
Response Code: 200
Response Reason: OK
ETag: None
2021-03-16 14:05:33,005 [update_manager.py:init_api_rate_limit()] - GitHub API Rate Limit Initialized
Rate Limit: 60
Rate Limit Remaining: 60
Rate Limit Reset Time: Tue Mar 16 15:05:32 2021, Seconds Since Epoch: 1615928732
2021-03-16 14:05:33,753 [shell_command.py:_check_proc_success()] - Command (apt list --upgradable) successfully finished
2021-03-16 14:05:33,754 [update_manager.py:refresh()] - Detected 0 package updates:

2021-03-16 14:05:33,771 [shell_command.py:_check_proc_success()] - Command (git -C /home/pi/moonraker status -u no) exited with return code 128
2021-03-16 14:05:33,772 [shell_command.py:run_with_response()] - Shell command 'git -C /home/pi/moonraker status -u no' output:
error: object file .git/objects/f0/9cd413ca47d8ee2ce93948fc50873c6b1afb27 is empty
error: object file .git/objects/f0/9cd413ca47d8ee2ce93948fc50873c6b1afb27 is empty
fatal: loose object f09cd413ca47d8ee2ce93948fc50873c6b1afb27 (stored in .git/objects/f0/9cd413ca47d8ee2ce93948fc50873c6b1afb27) is corrupt
On branch master

2021-03-16 14:05:34,286 [shell_command.py:_check_proc_success()] - Command (git -C /home/pi/moonraker status -u no) exited with return code 128
2021-03-16 14:05:34,287 [shell_command.py:run_with_response()] - Shell command 'git -C /home/pi/moonraker status -u no' output:
error: object file .git/objects/f0/9cd413ca47d8ee2ce93948fc50873c6b1afb27 is empty
error: object file .git/objects/f0/9cd413ca47d8ee2ce93948fc50873c6b1afb27 is empty
fatal: loose object f09cd413ca47d8ee2ce93948fc50873c6b1afb27 (stored in .git/objects/f0/9cd413ca47d8ee2ce93948fc50873c6b1afb27) is corrupt
On branch master

2021-03-16 14:05:34,813 [shell_command.py:_check_proc_success()] - Command (git -C /home/pi/moonraker status -u no) exited with return code 128
2021-03-16 14:05:34,814 [shell_command.py:run_with_response()] - Shell command 'git -C /home/pi/moonraker status -u no' output:
error: object file .git/objects/f0/9cd413ca47d8ee2ce93948fc50873c6b1afb27 is empty
error: object file .git/objects/f0/9cd413ca47d8ee2ce93948fc50873c6b1afb27 is empty
fatal: loose object f09cd413ca47d8ee2ce93948fc50873c6b1afb27 (stored in .git/objects/f0/9cd413ca47d8ee2ce93948fc50873c6b1afb27) is corrupt
On branch master

2021-03-16 14:05:35,340 [shell_command.py:_check_proc_success()] - Command (git -C /home/pi/moonraker status -u no) exited with return code 128
2021-03-16 14:05:35,341 [shell_command.py:run_with_response()] - Shell command 'git -C /home/pi/moonraker status -u no' output:
error: object file .git/objects/f0/9cd413ca47d8ee2ce93948fc50873c6b1afb27 is empty
error: object file .git/objects/f0/9cd413ca47d8ee2ce93948fc50873c6b1afb27 is empty
fatal: loose object f09cd413ca47d8ee2ce93948fc50873c6b1afb27 (stored in .git/objects/f0/9cd413ca47d8ee2ce93948fc50873c6b1afb27) is corrupt
On branch master

2021-03-16 14:05:35,868 [shell_command.py:_check_proc_success()] - Command (git -C /home/pi/moonraker status -u no) exited with return code 128
2021-03-16 14:05:35,868 [shell_command.py:run_with_response()] - Shell command 'git -C /home/pi/moonraker status -u no' output:
error: object file .git/objects/f0/9cd413ca47d8ee2ce93948fc50873c6b1afb27 is empty
error: object file .git/objects/f0/9cd413ca47d8ee2ce93948fc50873c6b1afb27 is empty
fatal: loose object f09cd413ca47d8ee2ce93948fc50873c6b1afb27 (stored in .git/objects/f0/9cd413ca47d8ee2ce93948fc50873c6b1afb27) is corrupt
On branch master

2021-03-16 14:05:36,370 [update_manager.py:initialize()] - Git Repo moonraker: Initialization failure
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/plugins/update_manager.py", line 748, in initialize
    self._verify_repo()
  File "/home/pi/moonraker/moonraker/plugins/update_manager.py", line 878, in _verify_repo
    f"Git Repo {self.alias}: repo not initialized")
utils.ServerError: Git Repo moonraker: repo not initialized
2021-03-16 14:05:36,371 [update_manager.py:refresh()] - Error Refreshing git state
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/plugins/update_manager.py", line 542, in refresh
    await self._update_repo_state()
  File "/home/pi/moonraker/moonraker/plugins/update_manager.py", line 549, in _update_repo_state
    await self.repo.initialize(need_fetch=need_fetch)
  File "/home/pi/moonraker/moonraker/plugins/update_manager.py", line 748, in initialize
    self._verify_repo()
  File "/home/pi/moonraker/moonraker/plugins/update_manager.py", line 878, in _verify_repo
    f"Git Repo {self.alias}: repo not initialized")
utils.ServerError: Git Repo moonraker: repo not initialized
2021-03-16 14:05:36,655 [update_manager.py:github_api_request()] - GitHub API Request Processed
URL: https://api.github.com/repos/meteyou/mainsail/releases/latest
Response Code: 200
Response Reason: OK
```

Arksine, 2021-03-17:
Thanks.  I have heard reports of this but this is the first time someone has been able to provide a log.

It isn't clear what is causing the corruption, most likely it is due to a git fetch or git pull failure.  I have made several attempts to mitigate this, but it seems like the only solution here is to use `rsync` to keep a backup of git repos and switch between them when a repo gets corrupted.

Arksine, 2021-03-18:
I just pushed a series of commits that will take a snapshot of a repo and attempt to recover a corrupt repo.

FWIW, after some research, a corrupt object suggest that there may be an issue with your filesystem.  Could be that the storage medium is going bad.  Could be a power supply issue.  I would recommend looking in to these things if corrupt repos are a common occurrence.

Arksine, 2021-04-20:
It should now be possible to recover.  In the future updates will likely move away from `git` in an effort to increase stability of the update manager.  I'm going to close this issue for now as in the interim it has been addressed.

closed 2021-04-20 by Arksine

jottr, 2021-12-09:
Please reopen:

```
2021-12-09 22:32:49,133 [websockets.py:add_websocket()] - New Websocket Added: 2806335600
2021-12-09 22:32:49,280 [websockets.py:build_error()] - JSON-RPC Request Error: 500
Update Manger Failed to Initialize
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/websockets.py", line 224, in execute_method
    result = await method(conn, *args, **kwargs)
  File "/home/pi/moonraker/moonraker/websockets.py", line 329, in func
    ip_addr=ws.ip_addr, user=ws.current_user))
  File "/home/pi/moonraker/moonraker/components/update_manager/update_manager.py", line 385, in _handle_status_request
    self._check_init_success()
  File "/home/pi/moonraker/moonraker/components/update_manager/update_manager.py", line 237, in _check_init_success
    raise self.server.error("Update Manger Failed to Initialize", 500)
utils.ServerError: Update Manger Failed to Initialize

```

electraflyit, 2021-12-11:
I have the same issue

Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/websockets.py", line 224, in execute_method
    result = await method(conn, *args, **kwargs)
  File "/home/pi/moonraker/moonraker/websockets.py", line 329, in func
    ip_addr=ws.ip_addr, user=ws.current_user))
  File "/home/pi/moonraker/moonraker/components/update_manager/update_manager.py", line 385, in _handle_status_request
    self._check_init_success()
  File "/home/pi/moonraker/moonraker/components/update_manager/update_manager.py", line 237, in _check_init_success
    raise self.server.error("Update Manger Failed to Initialize", 500)
utils.ServerError: Update Manger Failed to Initialize
2021-12-10 23:24:33,760 [websockets.py:build_error()] - JSON-RPC Request Error: 500
Update Manger Failed to Initialize
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/websockets.py", line 224, in execute_method
    result = await method(conn, *args, **kwargs)
  File "/home/pi/moonraker/moonraker/websockets.py", line 329, in func
    ip_addr=ws.ip_addr, user=ws.current_user))
  File "/home/pi/moonraker/moonraker/components/update_manager/update_manager.py", line 385, in _handle_status_request
    self._check_init_success()
  File "/home/pi/moonraker/moonraker/components/update_manager/update_manager.py", line 237, in _check_init_success
    raise self.server.error("Update Manger Failed to Initialize", 500)
utils.ServerError: Update Manger Failed to Initialize

Arksine, 2021-12-11:
Log snippets aren't useful for debugging, I need you to upload the full log.  FWIW, that error is no present, so its possible that manually updating will resolve this issue.

jottr, 2021-12-11:
Updated manually for now. I will report back with a full log if issue comes up again.

QGB, 2023-03-19:
```
[update_manager.py:initialize()] - PackageDeploy: Using PackageKit Provider
[update_manager.py:initialize()] -
Initializing Client Updater: 'mainsail',
Channel: stable
path: /home/ubuntu/mainsail
[update_manager.py:_handle_auto_refresh()] - update_manger: Install validation pending, bypassing initial refresh
[machine.py:_check_service_file()] - Performing Service Validation...
[server.py:add_warning()] - Unable to retrieve Moonraker service info.  Service file must be updated manually.
[extensions.py:start_unix_server()] - Creating Unix Domain Socket at '/home/ubuntu/printer_data/comms/moonraker.sock'
[server.py:start_server()] - Starting Moonraker on (0.0.0.0, 7125), Hostname: hi3798mv100
[app.py:listen()] - SSL Certificate/Key not configured, aborting HTTPS Server startup
[klippy_connection.py:_do_connect()] - Klippy Connection Established
[machine.py:extract_service_info()] - Error Extracting Service Info
Traceback (most recent call last):
  File "/home/ubuntu/moonraker-env/lib/python3.8/site-packages/dbus_next/proxy_object.py", line 217, in get_interface
    intr_interface = next(i for i in self.introspection.interfaces if i.name == name)
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/moonraker/moonraker/components/machine.py", line 1209, in extract_service_info
    unit_intf, svc_intf = await self.dbus_mgr.get_interfaces(
  File "/home/ubuntu/moonraker/moonraker/components/dbus_manager.py", line 125, in get_interfaces
    intf = proxy_obj.get_interface(ifname)
  File "/home/ubuntu/moonraker-env/lib/python3.8/site-packages/dbus_next/aio/proxy_object.py", line 159, in get_interface
    return super().get_interface(name)
  File "/home/ubuntu/moonraker-env/lib/python3.8/site-packages/dbus_next/proxy_object.py", line 219, in get_interface
    raise InterfaceNotFoundError(f'interface not found on this object: {name}')
dbus_next.errors.InterfaceNotFoundError: interface not found on this object: org.freedesktop.systemd1.Service
[server.py:add_log_rollover_item()] - Klipper Version: v0.11.0-123-g6b75f54e
[server.py:add_warning()] - file_manager: Klipper configuration file not located in 'config' folder.

Klipper Config Path: /home/ubuntu/klipper_config/printer.cfg
```