Arksine/moonraker #122: [FR] Controll WLED's / D1mini based on Print status / other events

itCarl, 2021-03-17:
**Led's go blink blink.**

Currently i have about 20 d1mini's around my house that run WLED to controll some leds and Lamps.
(and of course one on my 3D Printer that controlls a light bar)

**So here is my Plug-in Idea / FR:**
if a print fails (may be detected by some spaghetti detector) some defined lights go red or Blink or something.
if a print is done some defined lights go green or something.

[WLED ](https://github.com/Aircoookie/WLED)on Gtithub
[WLED json API ](https://github.com/Aircoookie/WLED/wiki/HTTP-request-API)

NovaViper, 2021-06-23:
I'd like to see this too! I don't want to have to setup Home Assistant in order to use these

richardjm, 2021-11-20:
I've added a wled module to moonraker. If you try it out and find it works (or doesn't) for you I'd be interested in knowing.
https://moonraker.readthedocs.io/en/latest/configuration/#wled

itCarl, 2021-11-21:
very nice, thx.
will try that in the course of the week.

closed 2021-12-22 by Arksine