Arksine/moonraker #126: Custom component; cached value

tobyworks, 2021-03-27:
Hello, im currently creating my own custom component and for this component i need a value that should be cached in between power cycles, it's a generated value.

i was think of having some kind of same approach as .moonraker_api_key but whenever i try to write a file it says permission denied.

Any help or advice would greatly be appreciated

dushyantahuja, 2021-03-27:
Klipper had a save variable command, maybe you can use that instead.

https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Reference.md#save_variables

Arksine, 2021-03-27:
It sounds like you don't have the appropriate permissions to write a file to the location you are choosing.

FWIW, I will very likely remove the api key file and have it persist in the database in the future.  You could add your persistent data there if you would like.  The `history` module provides an example of how to use the database, although I don't recommend creating a new namespace for a single value.  Use the existing `moonraker` namespace, then add the value to a key starting with the module.  Something like:
```python
database = self.server.lookup_component('database')
database.insert_item("moonraker", "module_name.key_desc", value)
my_val = database.get_item("moonraker", "module_name.key_desc")
```

tobyworks, 2021-03-27:
Hey Arkshine thanks for the response. I have indeed already figured out that a database entry is the way to go for me and it already works like a charm.

Will close this one!

closed 2021-03-27 by tobyworks