Arksine/moonraker #130: Created a Moonraker Plugin

2BWorks, 2021-03-29:
Hi there,

Love your work, for tour app 3D Geeks we have created a moonraker component that enabled users to get notified on printer progress and such.

Is this something your are interested in merging with your repo. if so, i will make sure to get it up to spec for PR.

If not i'll leave the repo as-is and send my users to the github repo for manual installation.

you can check out the component here:
https://github.com/2BWorks/Moonraker-3DGeeks-Component

Arksine, 2021-03-30:
Thanks, it looks good.  In the near future I plan to implement a 3rd party plugin interface, I think that this kind of plugin would be a good candidate for it.  Since it is tied to your app you may wish to be able to maintain it directly vs having to create a pull request when it needs a change or update.

My plan is to keep the 3rd party plugin API as similar to the existing component API as possible and deploy through the update manager.

2BWorks, 2021-03-31:
Is there already a experimental build on this with which i can test. i'd love to create a first third party plugin

Arksine, 2021-03-31:
No, I haven't implemented it yet.  Its still in the planning phase at the moment.