Arksine/moonraker #133: Power device API and "unRESTful" implementation

aschor, 2021-03-31:
Hi,

I have an issue, and don't find an answer in the docs.

I'm "simply" trying to control moonraker power devices from the API, but from "home assistant", with the help of [RESTful switches](https://www.home-assistant.io/integrations/switch.rest)

And I do not succeed, because the API's URL to call to switch an item is NOT THE SAME when you switch it ON or OFF (the on/off is included in the url, as well as the device). 
I expected to be able to send to ONE url (like "http://192.168.1.13:7125/machine/device_power/my_device" a json data of the form {"command":"on"}, but I can't.

I've seen the ["json-rpc"](https://moonraker.readthedocs.io/en/latest/web_api/#power-on-devices) request on the docs .... but for the life of me I cannot get it to work, I did not find to what endpoint I should send it :

this is what I've got : 
```
curl -X GET http://192.168.1.13:7125/machine/device_power/status?LEDs
{"result": {"LEDs": "off"}}
```

and here is what I've tried :
```
curl -X POST -H 'Content-Type: application/json' -d '{    "jsonrpc": "2.0",    "method": "machine.device_power.on",    "params": {        "LEDs":null    },    "id":4564}' http://192.168.1.13:7125
curl -X POST -H 'Content-Type: application/json' -d '{    "jsonrpc": "2.0",    "method": "machine.device_power.on",    "params": {        "LEDs":null    },    "id":4564}' http://192.168.1.13:7125/
curl -X POST -H 'Content-Type: application/json' -d '{    "jsonrpc": "2.0",    "method": "machine.device_power.on",    "params": {        "LEDs":null    },    "id":4564}' http://192.168.1.13:7125/machine/device_power/
curl -X POST -H 'Content-Type: application/json' -d '{    "jsonrpc": "2.0",    "method": "machine.device_power.on",    "params": {        "LEDs":null    },    "id":4564}' http://192.168.1.13:7125/
curl -X POST -H 'Content-Type: application/json' -d '{    "jsonrpc": "2.0",    "method": "machine.device_power.on",    "params": {        "LEDs":null    },    "id":4564}' ws://192.168.1.13:7125/
```

and they ALL fail with  :
```
<html><title>404: Not Found</title><body>404: Not Found</body></html>
```

Did I miss something ? It seems that with homeassistant's RESTful switches, the URL MUST be constant, and the data (at least the command) MUST be in the body (json) of the POST request ...

any help would be appreciated as for now I can't control the leds, but I have also security relays on hotend and heatbed and a powerplug after I get the leds working ...

Arksine, 2021-04-01:
This seems to be a limitation of the plugin you are using for Home Assistant.  If you want to toggle devices on/off via HTTP you must be able to request their individual endpoints.  I don't believe that this makes these API's "unrestful", it just doesn't conform to what your Home Assistant plugin expects.

FWIW, the JSON-RPC APIs aren't applicable to HTTP, they provide an alternative for communicating with Moonraker over a websocket.

aschor, 2021-04-01:
humm ok, so maybe what I'm describing is a missing API from my point of view ? a "(power) device api" ?

would it be difficult to separate the endpoint (uri) from the message (data as json) ? some sort of HTTP api that would read post json data as incomming message/command

I think that would be useful also for other things like sending gcode ; I mean : gcode should NOT be in the uri, especially with klipper. 

And in general, [clean urls](https://en.wikipedia.org/wiki/Clean_URL) (aka RESTful URLs, user-friendly URLs, pretty URLs) should separate the ressource path from the internal representation, in the case of power api, the "on" or "off" part in the uri should not be there .... and if we send a POST request, WHY should we send GET parameters with null data ? that's not usual

If I read [that from microsoft](https://docs.microsoft.com/fr-fr/azure/architecture/best-practices/api-design) (there is a "read in english switch in the upper right, I did not find the direct english link) :
this is a guideline :

> Focus on the business entities that the web API exposes. For example, in an e-commerce system, the primary entities might be customers and orders. Creating an order can be achieved by sending an HTTP POST request that contains the order information. The HTTP response indicates whether the order was placed successfully or not. When possible, resource URIs should be based on nouns (the resource) and not verbs (the operations on the resource).
HTTP
>
>https://adventure-works.com/orders // Good
>
>https://adventure-works.com/create-order // Avoid

so POST  http://host:7125//machine/device_power/on?  ......  this seems like a "verb" here

I know that's more than just my issue here, I'm discussing the REST aspect of the APIs, but I think it's worth it. For now, for my issue, I don't have any solution

maybe having an internal endpoint doing all the url rewriting could be some solution, without disturbing the actual APIs ? Can moonraker be augmented with plugins that could manage that ?

what do you think @Arksine ? Are you open on this "change" ?

Arksine, 2021-04-01:
Moonraker doesn't claim to adhere to a purely "RESTful" standard, in large part due to the way that Klipper can procedurally generate endpoints.  With regard to Microsoft's guidelines, its important to not that is all they are.  It isn't a specification that Moonraker's API claims to adhere to.

In my view, you have an issue on the client side.  That is where it needs to be resolved.

aschor, 2021-04-01:
ok, thanks for your input. Closing this issue then, and I'll try on homeassistant side

closed 2021-04-01 by aschor

Arksine, 2021-04-01:
Thanks.  FWIW, all of Moonraker's HTTP APIs can submit arguments via json in the body (for example, the gcode API can, so the script doesn't need to be in the URI).  And I do agree that using verbs on the power commands is a bit different, most of Moonraker's APIs don't do that.   I didn't catch it at the time I merged the pull request, and my concern is that changing it would break existing client APIs.  I may be open to a change at some point, but I don't want to break the API for existing clients at this time.

aschor, 2021-04-02:
thanks @Arksine 

I have solved (temporarily ?) my issue by creating a new "rest switch" for home assistant [here ](https://github.com/aschor/HASS_template_REST_switch)(if anyone is interested). It works beautifully 👯