Arksine/moonraker #134: Tasmota 1 relay powerdevice  not working

aschor, 2021-04-02:
Hi, me again.

I've found a bug in the code : 

Say I have a tasmota device, a sonoff S26 (that's what I have). this device has only 1 relay. The firmware is correctly flashed (I've tried on my "old one" and also on a fresh wemos d1 as a sonoff basic (the default after tasmota flash).

moonraker defaults output_id to "1", that is logical.

If I send a request to get the status, here is what happens :
```
arnaud@Arnaud-2650v3:~/core/config/custom_components$ curl -X GET "http://192.168.1.52/cm?user=admin&password=&cmnd=Power1"
{"POWER":"OFF"}
```

see the "get 'power1', receive 'power' " ?  you can argue it's a tasmota bug, but as long as the device has only one relay configured, the power device number is stripped out in the response (which is logical).

But moonraker [EXPECTS](https://github.com/Arksine/moonraker/blob/c36f989c529584753623eb14ab475511eeb5e0ef/moonraker/components/power.py#L458)  to see a number in the response .... so it does not get any status for the power device. no error is thrown, no log ... 

I think that here, moonraker SHOULD try to see if it's talking to a single relay device if it fails to get a multi-relay device status (that's actually what moonraker exepects by default)

Arksine, 2021-04-02:
Tasmota support was contributed by @0x4d4e.  Would you take a look at this?  Thanks.

0x4d4e, 2021-04-02:
Hi @aschor!

I noticed this as well with one of my devices during the original pull request of this code: https://github.com/Arksine/moonraker/pull/39#issuecomment-731571934

We decided back then, that including a fallback or attempt to detect this case was probably not necessary since it can be fixed by setting `SetOption26 1` in the Tasmota console.

It is also mentioned in the documentation for `output_id`, but could maybe be clearer:

    #   If your single-relay Tasmota device switches on/off successfully,
    #   but fails to report its state, ensure that 'SetOption26' is set in
    #   Tasmota.

Please check if this solves this issue for you.

aschor, 2021-04-02:
hi @0x4d4e 

no, it won't work ; in tasmota :
```
11:54:28.906 CMD: SetOption26
11:54:28.911 RSL: stat/tasmota_53242F/RESULT = {"SetOption26":"ON"}
```

on command line after a reset of the tasmota and a check that this option is on : 
```
curl -X GET "http://192.168.1.52/cm?user=admin&password=&cmnd=Power1"
{"POWER":"OFF"}
```

I've managed to add the necessary code (that easy to fix), I'll try to do a PR (that will be a 1st for me )

0x4d4e, 2021-04-02:
Oh well, then a fallback check if probably necessary. 

Great @ pull request! The one I linked has a detailed explanation by Arksine on how to sign your commits.

aschor, 2021-04-02:
PR done, let me know if I've done things right :)

PR #135 (2021-04-02): power.py: fix issue with single relay tasmota
add a check and handle consequently in case tasmota device is only one relay and would return a power value when asked a power1 value

Fixes #134
closed 2021-04-02 by Arksine