Arksine/moonraker #136: The gpio control does not work

sashadeg, 2021-04-03:
My relay is connected to gpio1. Raspberry pi 3b
I am adding code to file moonraker.conf based on example.

[power printer]
type: gpio
pin: gpio1
off_when_shutdown: True
initial_state: off

A switch appeared in the fluid, but it doesn't work. The state of the gpio does not change. 
Controlling gpio from ssh work fine. (gpio mode 1 out, gpio write 1 1)

Later I installed this package (sudo apt-get install wiringpi) but it didn't help.
I reinstalled fluidPI but it didn't help.

p.s. I was able to control the "tasmota" device =)
But the gpio doesn't work in any way.
[moonraker (2).log](https://github.com/Arksine/moonraker/files/6253301/moonraker.2.log)

Arksine, 2021-04-03:
Most likely you have something on your system that is conflicting with libgpiod.  WiringPi may do this.  Moonraker must be able to gain exclusive control of the pin.

Just to be sure, I have verified that GPIO1 on my system will correctly toggle a LED on and off.

sashadeg, 2021-04-04:
> WiringPi may do this. Moonraker must be able to gain exclusive control of the pin.
How can I check this? I am working with a clean system. Why does not it work? Maybe I need to set some permissions?

sashadeg, 2021-04-04:
I installed klipper, moonraker and fluid on clear system:
1. Using a script "kiauh". 2019-09-26-raspbian-buster.
2. Using a script "kiauh". 2020-02-13-raspbian-buster
3. Using a script "kiauh". Raspberry Pi OS Lite Release date: March 4th 2021
4. Using a script "kiauh". Raspberry Pi OS with desktop and recommended software March 4th 2021
5. Manually installed packages. 2019-09-26-raspbian-buster.
6. Manually installed packages. Raspberry Pi OS Lite Release date: March 4th 2021
7. Installed FluiddPi.
I'm already going crazy! The gpio control from fluid does not work anywhere! But manually using ssh everything works! 
Help me please!

Arksine, 2021-04-04:
I will try.  First, you need to make absolutely sure nothing else is trying to control that pin.  Do NOT install WiringPi.  Do not install anything but Klipper and Moonraker.

Next, enable debug logging in by adding the follwing option to the [server] section in moonraker.conf: ```
[server]
enable_debug_logging: True
```
Restart the moonraker service:
```
sudo service moonraker restart
```
Now connect with Fluidd and try to toggle the pin a few times.  Then download `moonraker.log` and post it here.

Arksine, 2021-04-04:
One other thing...WiringPi may use a different pin numbering scheme than libgpiod.  Make sure your relay is actually connected to GPIO1 as noted on [this pinout](https://pinout.xyz/).

sashadeg, 2021-04-04:
[moonraker (3).log](https://github.com/Arksine/moonraker/files/6254959/moonraker.3.log)
Installed clean FluiddPi. I didn't install anything else. Pushing the switch. I drop the log.
p.s. Without installing WiringPi, the commands for controlling the gpio from ssh do not work.

sashadeg, 2021-04-04:
> Еще одна вещь ... WiringPi может использовать другую схему нумерации контактов, чем libgpiod. Убедитесь, что ваше реле действительно подключено к GPIO1, как указано на [этой распиновке](https://pinout.xyz/) .

OMG! Its working! I watched the pinout diagram from "gpio readall". It does not match the scheme from pinout.xyz
Gpio1 corresponds to gpio18
[image]
[image]

closed 2021-04-20 by Arksine