Arksine/moonraker #138: Implement notification component

ayufan, 2021-04-05:
I would like to write a notification component that would allow to trigger a set of actions on a different events happening in Klipper.

For that, I would like to:
- abstract out of history component the `print_stats` state tracking
- rework `history` to use a common base interface
- implement notification component that would implement actions similar to: `on_print_start/on_print_resume/pause`
- notification would have many pluggable backends (similar to power) allowing to use http or gcode macros initially
- the common base interface could also be used by another components, like timelapse to consistently track a print state

Is this a good direction?

ayufan, 2021-04-05:
@Arksine 

WDYT? Is this a good direction? I'm in particular asking about abstracting the state tracking which is likely to be reimplemented by many components even though it should be pretty common set of behaviors.

Arksine, 2021-04-05:
I think the general idea is a good one, I have plans to add support for MQTT and I could simply interface with your component rather than implement from scratch.

With regard to the interface, I would say that "it depends".  It might make sense to create an interface (or a helper) that tracks `print_stats` and reports various state transitions.   Beyond that, what each component does with this is going to be different.  The question  becomes as to whether or not there is enough code reuse to matter.

I think a good approach would be to create the interface, have your component use the interface, then at a later date we could look at modifying `history`.  I don't think we would want to do both in the same pull request.

ayufan, 2021-04-05:
@Arksine 

I was thinking more about 2-4 PRs :)

Arksine, 2022-07-09:
Since Moonraker now has a notifier component (along with MQTT) I think we can mark this issue as resolved.  That said, there is probably room to expand the notifier to support more events.

closed 2022-07-09 by Arksine