Arksine/moonraker #140: Tuya Smartplug integration

morganfw, 2021-04-06:
Hello,
is it planned or is it possible in the future to integrate the Tuya Smartplug plugin?
Here are the links to the integrations on OctoPrint for example: https://plugins.octoprint.org/plugins/tuyasmartplug/ and https://github.com/ziirish/OctoPrint-TuyaSmartplug

Regards

sickerin, 2023-02-04:
Did anyone manage to find a work around, if it's not available yet? Currently, what I am considering to do is install homeassistant, then add the Tuya plugs to it. From there I presume moonraker can add homeassistant. Not sure if this will work.

fer662, 2024-04-29:
Well, hopefully someone still wants this. https://github.com/Arksine/moonraker/pull/846
[image]