Arksine/moonraker #141: Node Modules update doesnt work

eliteSchwein, 2021-04-08:
after the big mooncord update it doesnt start anymore because moonraker doesnt install the new required node modules

[image]
2021-04-08 23:54:38,688 [shell_command.py:_check_proc_success()] - Command (git -C /home/pi/mooncord pull --progress) successfully finished
2021-04-08 23:54:38,690 [update_manager.py:_log_info()] - Repo MoonCord: No packages found in script: /home/pi/mooncord/scripts/install.sh
2021-04-08 23:54:38,715 [shell_command.py:_check_proc_success()] - Command (git -C /home/pi/mooncord status -u no) successfully finished
2021-04-08 23:54:38,728 [shell_command.py:_check_proc_success()] - Command (git -C /home/pi/mooncord config --get branch.master.remote) successfully finished
2021-04-08 23:54:38,743 [shell_command.py:_check_proc_success()] - Command (git -C /home/pi/mooncord branch --list) successfully finished
2021-04-08 23:54:38,757 [shell_command.py:_check_proc_success()] - Command (git -C /home/pi/mooncord remote get-url origin) successfully finished
2021-04-08 23:54:38,771 [shell_command.py:_check_proc_success()] - Command (git -C /home/pi/mooncord rev-parse HEAD) successfully finished
2021-04-08 23:54:38,785 [shell_command.py:_check_proc_success()] - Command (git -C /home/pi/mooncord rev-parse origin/master) successfully finished
2021-04-08 23:54:38,810 [shell_command.py:_check_proc_success()] - Command (git -C /home/pi/mooncord describe --always --tags --long --dirty) successfully finished
2021-04-08 23:54:38,826 [shell_command.py:_check_proc_success()] - Command (git -C /home/pi/mooncord describe origin/master --always --tags --long) successfully finished
2021-04-08 23:54:38,826 [update_manager.py:log_repo_info()] - Git Repo MoonCord Detected:
Owner: eliteSchwein
Path: /home/pi/mooncord
Remote: origin
Branch: master
Remote URL: https://github.com/eliteSchwein/mooncord.git
Current Commit SHA: cf85c7013ad4c68658d8ed1c41edd85ce442c14a
Upstream Commit SHA: cf85c7013ad4c68658d8ed1c41edd85ce442c14a
Current Version: v0.0.2-0
Upstream Version: v0.0.2-0
Is Dirty: False
Is Detached: False
Commits Behind: 0
2021-04-08 23:54:39,053 [shell_command.py:_check_proc_success()] - Command (rsync -a --delete /home/pi/mooncord/ /home/pi/.mooncord_repo_backup) successfully finished
2021-04-08 23:54:39,054 [update_manager.py:_log_info()] - Repo MoonCord: Validity check for git repo passed
2021-04-08 23:54:39,118 [websockets.py:remove_websocket()] - Websocket Removed: 2817807312
2021-04-08 23:54:39,123 [shell_command.py:_check_proc_success()] - Command (sudo systemctl restart MoonCord) successfully finished

Arksine, 2021-04-08:
Can you provide the full log?  Thanks.

eliteSchwein, 2021-04-11:
Works as expected

closed 2021-04-11 by eliteSchwein