Arksine/moonraker #144: Power plugin does not detect device being on

ayufan, 2021-04-14:
My current configuration is to have TP-Link power switch connected to board,
and Raspberry PI 4 connected separately (always-on).

Sometimes I might power on device with a physical switch. Moonraker in such
cases does not detect power switch change of state. The downside of such
is that since I have a `idle_timeout` configured to automatically turn off power
this is ignored when executed by Klipper.

I think ideally, Moonraker should:

- either pull a state of a switch (likely preferred)
- assume that power switch is ON when klippy is connected

Arksine, 2021-04-14:
I could look into refreshing the state when Klippy is connected, however I don't want to poll the device.  Making assumptions about the switch state would break other usage scenarios.  Ideally Moonraker has complete control over you switch.

In the meantime you could get around this by calling the `set_device_power` remote method from a delayed gcode when Klipper enters the ready state.

ayufan, 2021-04-14:
> I could look into refreshing the state when Klippy is connected, however I don't want to poll the device 

I think it is reasonable to pool the device. There are various reasons, one for example being able to read energy readings available on some devices. This would ensure that switch state is reflected. Whether the switch is enabled by user (clicking UI) or it is read to be switched on seems like the outcome is very similar.

> Ideally Moonraker has complete control over you switch.

This is not entirely true, as you can still enable switch with a button :)

> In the meantime you could get around this by calling the set_device_power remote method from a delayed gcode when Klipper enters the ready state.

Interesting hint.

onfoot, 2021-04-21:
At least with Shellies we can define a URL the device calls when the relay state is changed, so we can set moonraker's endpoint there to keep the state up-to-date.

Don't know about TP-Link things though.

I agree it's unrealistic to assume klipper is exclusively in control of the switch. It should be, sure, because dropping that assumption makes the `locked_when_printing` option useless, for example, and we should probably have a way to prevent powering the printer off while printing (say by shouting at Siri to turn off all the devices 🙈).

Arksine, 2021-04-22:
> At least with Shellies we can define a URL the device calls when the relay state is changed, so we can set moonraker's endpoint there to keep the state up-to-date.

This would be preferable to polling, however I doubt that every device would support it.  I'll look into the TP Link devices.  If this seems widely available I'm open to providing an endpoint to update the status.  TBH I would like to refactor the power APIs so a single path can toggle device power, I just don't want to break client functionality at the moment.

> I agree it's unrealistic to assume klipper is exclusively in control of the switch.

Moonraker's assumption of its control is exclusive to the additional functionality it applies when commanded to toggle device power.  As you say, it can prevent and inadvertent power off when it is in control.  Likewise it can do things like restart the klipper service after a power on (which is what the OP is requesting).  However it is my view that Moonraker should only do this when commanded to do so, I don't believe it would correct for Moonraker to take action based on an change of device state.  I fear that this could lead to unexpected outcomes.

ayufan, 2021-04-22:
> However it is my view that Moonraker should only do this when commanded to do so, I don't believe it would correct for Moonraker to take action based on an change of device state. I fear that this could lead to unexpected outcomes.

I fully agree with that. It might also be controllable, but it would be desired to ensure that state is reflected.

Arksine, 2021-05-10:
I have pushed an update that always refreshes the device state before running a command.  This should prevent Moonraker from taking an inadvertent action should the internal device state be stale.

Arksine, 2021-06-18:
I'm going to go ahead and close this, as Moonraker does now attempt to verify state before taking an action.

closed 2021-06-18 by Arksine