Arksine/moonraker #145: Retain print history when moving files through the API

kageurufu, 2021-04-15:
I started a reasonably long series of prints, then decided I wanted all the jobs in a single folder to keep it organized instead of just at the root of the virtual SD.

When I moved the files in Fluidd, it lost the history when the files were moved

Arksine, 2021-04-15:
The history should still be there, however it is no longer associated with your gcode files since they have been moved. 
 History items themselves are not mutable.  The location of the file is part of the history and thus cannot be changed.

Arksine, 2021-04-20:
I'm going to go ahead and close this as it isn't a bug and the behavior is intentional (the history is not mutable).

closed 2021-04-20 by Arksine