Arksine/moonraker #148: CORS configuration with the Kiri:Moto slicer and missing X-Api-Key

davidjericho, 2021-04-18:
Hello,

I was setting up the Kiri:Moto slicer (https://grid.space/kiri) to push to Moonraker, and setup the Octoprint integration via the CORS configuration.

Talking with Stewart of the GridSpace project, we found that browsers were blocking the Octoprint integration because the X-Api-Key header was being sent regardless of it being set or not, and the browsers were rejecting the CORS request because of the missing header. 

He has modified the behaviour in the Kiri:Moto development branch so if no key has been set, no X-Api-Key header is sent, however this doesn't solve the strong authentication case. 

The Firefox console was reporting to me:

`Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at https://xxxxx/api/files/local. (Reason: header ‘x-api-key’ is not allowed according to header ‘Access-Control-Allow-Headers’ from CORS preflight response).`

Modifying https://github.com/Arksine/moonraker/blob/e8f6862fd26c45551959835378a08295e0538667/moonraker/authorization.py#L235 to add X-Api-Key as a permitted header, permits the browser to do the Octoprint API upload with CORS.

Arksine, 2021-04-18:
Ok.  At the moment the Auth module is under heavy development as I am adding JWT Authorization support.  When I push that I will update the headers to allow X-Api-Key through.

Arksine, 2021-04-20:
This should be resolved by commit 4e0e2235990fbdb2d1e75e664f133ac9b51fc9cb.

davidjericho, 2021-04-21:
Thank you @Arksine, I can confirm that this works as expected.

closed 2021-04-21 by davidjericho