Arksine/moonraker #150: Feature Request - GPIO buttons

doe57716, 2021-04-20:
Is it possible to add GPIO buttons for switching power devices on/off? such as a light or the printer

Arksine, 2021-04-20:
Yes, the [power](https://moonraker.readthedocs.io/en/latest/configuration/#power) component can be configured to support GPIOs.

doe57716, 2021-04-21:
I already have the relay connected to gpio configured and working from the web interface, and want to add a physical switch to toggle them, like octoprint enclosure

Arksine, 2021-04-21:
Ah, I see.  Moonraker does not support reading GPIOs.  That encroaches a bit on Klipper's functionality, so I'm unsure if it is a good idea to implement it.  A proper solution to this would be to have something like an esp8266 monitor some GPIOs and toggle the API when a button is pressed (or released).  This way a button could be used to do virtually anything.

samwiseg0, 2021-10-14:
I too would like to see this. My idea would be that a power device can have an associated button to control it via GPIO. 

example:
```
[power MCU]
type: tasmota
toggle_pin: gpio17
locked_while_printing: True
address: 192.168.x.xxx
restart_klipper_when_powered: True
restart_delay: 5.
```

LazaroFilm, 2022-06-18:
Adding my voice to this thread. This would really help a lot. Not having to use A screen to turn the printer on would be really nice.

Arksine, 2022-06-18:
FWIW, Moonraker now has limited support for [GPIO Buttons](https://moonraker.readthedocs.io/en/latest/configuration/#button)

closed 2022-07-09 by Arksine