Arksine/moonraker #151: TPLink HS110 no longer works due to firmware update (Hardware 4.1, but will eventually effect all plugs)

nirurin, 2021-04-20:
https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/issues/218

Same issue as here, but it affects moonraker too. All TPLink devices will be changing over to the 'new API' soon, but the ones that have already been altered no longer work with existing controllers like the Octoprint plugin or the moonraker implementation. 

I can only confirm that it affects HS110 plugs with hardware version 4.1, which are on firmware 1.10. But I believe people have had the same issue on other plugs, and TPLink have confirmed that its a new API implementation so it will go out to all plugs. 

Hopefully someone will look into this and implement the new API, cos power control from within moonraker is really useful and important.

Arksine, 2021-04-22:
If TP Link is forthcoming with its API documentation then I have no issues adding support for it.  I'm not keen on being forced to register a developer account with them just to get access to their API.

DanielJoyce, 2021-06-13:
python-kasa master can successfully communicate with tp-link hubs, hardware 5.0

```
❯ kasa
No host name given, trying discovery..
Discovering devices on 255.255.255.255 for 3 seconds
== ender5 plug - HS103(US) ==
	Host: 192.168.86.121
	Device state: ON

	== Generic information ==
	Time:         2021-06-12 19:04:58
	Hardware:     5.0
	Software:     1.0.3 Build 201015 Rel.142523
	MAC (rssi):   90:9A:4A:D9:9E:31 (-68)
	Location:     {'latitude': 479644, 'longitude': -1222213}

	== Device specific information ==
	LED state: True
	On since: 2021-06-12 18:17:40.542175
```

Arksine, 2021-12-22:
It doesn't look like TPLink is interested in providing a viable alternative in the near term.  According to [this Home Assistant alert](https://alerts.home-assistant.io/#tplink.markdown) it seems like they even stopped providing Beta Firmware that re-enables the local API.

I have added a warning to Moonraker's documentation that some devices may not work.  For the time being I think the best course of action is to close this issue, as it cannot be resolved until TPLink releases its new local API.

closed 2021-12-22 by Arksine