Arksine/moonraker #152: Cancel/Pause/Resume in Octoprint Compat

TheStealthReporter, 2021-04-21:
I've noticed that the cancel/pause/resume functionality is not implemented in the octoprint compat module (always returns a "405 Not Allowed", because POST requests are not supported). The `/api/job` endpoint would have to parse the POST request for example:

```
{"command": "cancel"}
```
(other operations are documented in https://docs.octoprint.org/en/master/api/job.html)

TheStealthReporter, 2021-04-30:
For my setting the most important command would be "cancel". A fast way to cancel a print when things go wrong is very important. Pause/resume is neat too but I rarely use it :)

I'd say for Cura start is not that important, since Cura will always upload the gcode, even if it already exists on the printer.

Arksine, 2021-05-10:
FWIW, the primary goal of Octoprint Compat is to support slicer uploads.  I dont think that the goal should be to support a wide range of Octoprint APIs.  IMO for direct control a Cura plugin developer should implement a plugin that uses Moonraker's native APIs.  I feel like extending support in `octoprint_compat` would discourage this.

grigi, 2021-05-10:
That is a valid point. The mere presence of octoprint compatibility will result in us doing a catch-up all the time to the octoprint API.

I was considering wether I should make the `octoprint_compat` plugin allow things like the Android OctoApp to work, but in two minds about it.
Some of the API's would map cleanly enough, but some (like select/start print/ reading file access) would not really work well.

Ensuring that it works for all the popular slicers is important, but we may already be in the position where we discourage slicers implementing moonraker plugins. :pensive:

swalsh11, 2021-05-12:
I have started using the octoprint compatibility, previously I was only using emtraxltd cura2moonraker plugin to upload files and start the print. 

https://github.com/emtrax-ltd/Cura2MoonrakerPlugin

Both work nicely side by side, Is there a way to merge the start print functionality?

grigi, 2021-05-12:
The api's don't map 1:1

## Moonraker:
1. Upload file `foo.gcode`
2. Start print on file `foo.gcode`

## Octoprint
1. Upload file `foo.gcode`
2. Select file `foo.gcode`
3. Print selected file

Not that difficult, tbh.

Didn't know about the cura moonraker plugin. Would be great if it could be part of the official plugin registry.

grigi, 2021-05-12:
I have implemented the job command api, but starting won't work yet: https://github.com/grigi/moonraker
Also note that I haven't tested this yet to see that it behaves exactly to spec, and that the real octoprint does the same.

You can test it in the meantime by just copying https://github.com/grigi/moonraker/blob/master/moonraker/components/octoprint_compat.py to your local version and restart moonraker.

swalsh11, 2021-05-12:
I will test it at the weekend, Thank you

jonferreira, 2021-08-30:
@grigi any more updates on this subject?

Was reading octoprints' API and there's a way to select and print a file all at once with POST /api/files/local/{file}?print=true

Presumably that's now how Curas plugin is doing though and thus the extra step needed to implement file select?

guigashm, 2021-12-20:
Hello,

First thank you for the great work.. It is almost perfect..   But  ..  :D  We miss a lot lot the "Auto start print immediately " that the octoprint cura plugin, has as an option.. 

I have the it active by setting the flag  in the config Octoprint plugin  menu in Cura  .. 
but is not implemented in the Moonracker Octoprint_compact .. like it is in the cura2moonraker plugin  ( by  emtraxltd ) that is able to upload files and start the print immediately .  ( https://github.com/emtrax-ltd/Cura2MoonrakerPlugin )

So looks like is possible to do,  because  , Emtraxltd  implemented this in the Cura2moonraker plugin .

But I prefer...  a LOT ...  :DDD ....   the  Octoprint_compact , so it would be great to press the button in Cura (as it is today)  .. but without having to go to the Mainsail or Fluid and have to go to the files and select and start printing.. 

Please.. Please .. Pleeeeease..  can you help us.. :D

By the way .. To test the changes you did, I tried to copy the content above -> https://github.com/grigi/moonraker/blob/master/moonraker/components/octoprint_compat.py

But an error is reported by Moonraker in  Mainsail, saying it found an error uploading the component Octopi_Compat..  Send the log in attach. 
[moonraker.log](https://github.com/Arksine/moonraker/files/7746702/moonraker.log)

Arksine, 2021-12-22:
Upload and Print should now be functional for Cura.  Moonraker has always supported the "start" flag, however Cura's plugin does not set when UFP file uploads are enabled.  Presumably this is because Octoprint would attempt to start the print before the  UFP plugin is finished extracting the gcode file.  Cura's Octoprint plugin gets around this issue by waiting for the UFP upload request to return, then it calls Octoprint's file command endpoint. I added support for Octoprint's file select and print functionality.  

I also added an `enable_ufp` option to `[octoprint_compat]`.  It will default to True, however it might be worth disabled.  When UFP is not enabled, Cura's Octoprint plugin sets the start flag when uploading.

All of that said, there is no guarantee that this wont break in the future, which was always my primary concern with attempting to support additional Octoprint APIs.  Given the maintenance burden I am going to restrict the Octoprint compatibility layer to supporting file uploads.  I won't be add or accept changes to support additional APIs such as pause, cancel, etc.  If users desire more control from Cura it will need to be done through a dedicated Cura plugin for Moonraker.  As such, I'm going to close this issue as "resolved" for file uploads, and "wont-fix" for the request support Cancel/Pause/Resume APIs.

closed 2021-12-22 by Arksine