Arksine/moonraker #153: From SuperSlicer no Upload possible.

Elektrickser, 2021-04-22:
After the last update, i can't make an File Upload to Moonraker.

And when i make a test in Printer-Configuration, it come the same failure:
`Fehler beim Hochloden zu Druckhost:
HTTP 401: {"error": {"code": 401, "message": "Unauthorized", "traceback": "Traceback (most recent call last):\n\n File \"/home/pi/moonraker-env/lib/python3.7/site-packages/tornado/web.py\", line 1681, in _execute\n result = self.prepare()\n\n File \"/home/pi/moonraker/moonraker/app.py\", line 283, in prepare\n self.current_user = auth.check_authorized(self.request)\n\n File \"/home/pi/moonraker/moonraker/components/authorization.py\", line 445, in check_authorized\n raise HTTPError(401, \"Unauthorized\")\n\ntornado.web.HTTPError: HTTP 401: Unauthorized (Unauthorized)\n"}}`

Moonraker log is here:
[moonraker.log](https://github.com/Arksine/moonraker/files/6360811/moonraker.log)

Arksine, 2021-04-22:
I'm not seeing those errors in your log, but I suspect that you have one of the following issues:

* The machine your are running superslicer from a PC that is not in a trusted IP range

* You are using an API Key for superslicer and it is incorrect.  The latest update generated a new API Key.  It is now stored in the database.  There is a script you can run to fetch the API Key, or you can get it by navigating to `http://<hostname>/access/api_key`, where `<hostname>` is name of your Moonraker host.   See the [user changes doc](https://moonraker.readthedocs.io/en/latest/user_changes/#april-19th-2021) for details.

As long as one of the above conditions are met you should be able to upload form superslicer.

Elektrickser, 2021-04-26:
Ok Thanks, it was the second choice. Now works fine.

closed 2021-04-26 by Elektrickser