Arksine/moonraker #155: Feature Request: Use of hostnames in trusted_clients

tjschweizer, 2021-04-29:
Hello,

Forgive me if this isn't possible, I'm not too familiar with networking protocols, but I'd like to be able to use some hostnames in the trusted_clients part of the [authorization] section in the config. 

I work on a campus and all of our computers are assigned hostnames but they don't have static IP addresses, and I'd like to restrict access to the printers to a specific group of computers. For example if I could make a config section like:

trusted_clients:
 example_computer1.example.domain
 example_computer2.example.domain

etc. 

I've been playing around with it and I think I could just put something like this in the moonraker/components/authorization.py file

```python
import socket 
...
for ip in trusted_clients:
            # Check IP address
            try:
                #add some logic to determine if entry is hostname or ip address, 
                #perhaps an entirely different section type needs to be created, I'm not sure
                if(is_hostname):
                      tc=ipaddress.ip_address(socket.gethostbyname(ip))
                else:
                      tc = ipaddress.ip_address(ip)
               
```

Is this something that is doable, or am I approaching this incorrectly?

Arksine, 2021-04-29:
This seems useful and easy to implement, so I added it in commit 46e1d7b.  I handled it a bit differently, where I look up the domain name via the IP received in the request header.  This way if dhcp changes the ip address of you clients Moonraker should still be able to authorize them without a restart.

Arksine, 2021-04-29:
As an aside, Moonraker also supports token authorization using JSON Web Tokens.  Clients will likely implement this in the near future, so it will be possible to create authorized users for your machines and entirely forego the `trusted_clients` option if you prefer.