Arksine/moonraker #157: Using the Fluidd UI and moonraker, comms to the Pi get blocked all the time

vgroenewold, 2021-05-05:
This seems like a buffer issue, but I often (few times a day) have the problem that I execute something on the Fluidd UI (like bed leveling, or adjusting Z while printing, or... seems rather random) and the UI becomes unresponsive. I can still click around, but commands are not being send to the Pi nor received, prints always continue so my guess is that the Pi becomes too busy. But how can I diagnose that or even solve it?

edit: I did some extra diagnosis with Htop running at the same time in a shell and that simply freezes as well for instance when the print starts for heating up and leveling. No comms whatsoever during that time, is it moonraker blocking that maybe? Resources looked totally fine. The only way to get responsiveness back immediately it so disconnect the Pi from the printer.

Thanks

Arksine, 2021-05-05:
This sounds like a Kernel or driver issue.  Even at 100% load Moonraker shouldn't be able to freeze ssh on a multicore system.

If you can reproduce this, disconnect the printer, then grab `/tmp/moonraker.log` (give it 30 seconds or so) it might provide some insight.  When Klippy enters the shutdown state Moonraker dumps its own process usage stats to the log.

vgroenewold, 2021-05-05:
Thanks, I'll try to recreate it and see if I can get a relevant section from the log.

vgroenewold, 2021-05-05:
Mm, nothing really special, I do see a wait command: 2021-05-05 19:08:04,006 [moonraker.py:wait()] - Request 'gcode/script' pending: 60.00 seconds

That was about the time it was not responding this time. I'll add more logs to this reply when I investigate further.

Arksine, 2021-05-05:
I'll need the full log to be able to help diagnose any potential issues.  Thanks.

vgroenewold, 2021-05-05:
Ok, sorry for that, let me finish a print and then I'll try to force the error and send the entire log. Thanks for the help!

vgroenewold, 2021-05-06:
Sorry for the long file, but this is the one from today. I at least experience 2 "disconnects" just during printing. I'm also wondering, even though my Pi has good wifi, if it may be a connection problem there.

[klippy.log](https://github.com/Arksine/moonraker/files/6436682/klippy.log)

Arksine, 2021-05-06:
Thanks, however `moonraker.log` is the one I require.  I might be able to glean some info from Klippy.log, but if there is a specific issue with moonraker it will be in there.

vgroenewold, 2021-05-06:
Oh wow, must have been a long day today. Let me try that again tomorrow, sorry about that.

vgroenewold, 2021-05-07:
Alright, here is the new one. During the print I saw Klippy not updating for a big period towards the end;
[moonraker.log](https://github.com/Arksine/moonraker/files/6443844/moonraker.log)

Arksine, 2021-05-07:
Looks like the websocket connect was dropped.  Since it occured in the middle of a status update some exceptions are logged while it tries to write to a closed socket, which is to be expected.

My guess is that its some kind of networking issue.

vgroenewold, 2021-05-07:
Ok, interesting. My pi is connected to wifi which is strong here, but maybe there's an issue there. I might try to hook it up via cable to check. Thanks!

closed 2021-05-13 by vgroenewold

QGB, 2023-03-07:
moonraker How to execute gcode wait to finish blocking call?