Arksine/moonraker #158: Feature Request: Support Hue Outlet using Hue Api

camatthew88, 2021-05-06:
Using a curl command we can toggle hue outlets/lights on and off. This would be useful to those already having a hue network and since the outlet api rarely if ever changes it will be fairly stable. Here is an example of curl commands used for psu control. https://community.octoprint.org/t/get-octoprint-psucontrol-running-with-philips-hue-smartplug/19057

olli-t, 2022-07-28:
Hi guys, 

I implemented the hue socket as power device, so it should work now, after configuring ;) . Please consider leaving feedback or problems regarding the current implementation.

terrorhai, 2022-09-01:
Hey, that sounds awesome! I will try it as soon as possible.

Experte9, 2023-01-12:
> Hey, that sounds awesome! I will try it as soon as possible.

And, does it work?

Experte9, 2023-04-04:
> The documentation doesn't seem complete: https://moonraker.readthedocs.io/en/stable/configuration/#hue-device-configuration. ...However I made it work using this:
> 
> [power Printer] type: hue address: 192.168.100.100 user: ************* device_id: 13 device_type: light

Correct, the documentation is not complete. For example my Code looks like this:
```
[power XXXXXXX]
type: hue
#   The type of device.  Can be either gpio, klipper_device, rf,
#   tplink_smartplug, tasmota, shelly, homeseer, homeassistant, loxonev1,
#   smartthings, mqtt or hue.
#   This parameter must be provided.
off_when_shutdown: True
#   If set to True the device will be powered off when Klipper enters
#   the "shutdown" state.  This option applies to all device types.
#   The default is False.
off_when_shutdown_delay: 300
#   If "off_when_shutdown" is set, this option specifies the amount of time
#   (in seconds) to wait before turning the device off. Default is 0 seconds.
on_when_job_queued: False
#   If set to True the device will power on if a job is queued while the
#   device is off.  This allows for an automated "upload, power on, and
#   print" approach directly from the slicer, see the configuration example
#   below for details. The default is False.
locked_while_printing: True
#   If True, locks the device so that the power cannot be changed while the
#   printer is printing. This is useful to avert an accidental shutdown to
#   the printer's power.  The default is False.
restart_klipper_when_powered: True
#   If set to True, Moonraker will schedule a "FIRMWARE_RESTART" to command
#   after the device has been powered on. If it isn't possible to immediately
#   schedule a firmware restart (ie: Klippy is disconnected), the restart
#   will be postponed until Klippy reconnects and reports that startup is
#   complete.  Prior to scheduling the restart command the power device will
#   always check Klippy's state.  If Klippy reports that it is "ready", the
#   FIRMWARE_RESTART will be aborted as unnecessary.
#   The default is False.
restart_delay:1
#   If "restart_klipper_when_powered" is set, this option specifies the amount
#   of time (in seconds) to delay the restart.  Default is 1 second.

address:192.168.XXX.XXX
#   A valid ip address or hostname of the Philips Hue Bridge. This
#   parameter must be provided.
user:XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
#   The api key used for request authorization.  This option accepts
#   Jinja2 Templates, see the [secrets] section for details.
#   An explanation how to get the api key can be found here:
#   https://developers.meethue.com/develop/get-started-2/#so-lets-get-started
device_id:XX
#   The device id of the light/socket you want to control.
#   An explanation on how you could get the device id, can be found here:
#   https://developers.meethue.com/develop/get-started-2/#turning-a-light-on-and-off
```

Experte9, 2024-01-13:
> Has anyone gotten this to work, and if so would you mind sharing your end-to-end configuration and execution code (excluding your private api key of course)?
> 
> My goal is to create a klipper command/macro to turn on a filtration fan (via a Hue smart plug) when the machine is actively printing something and then turn it off shortly after printing stops.

I can only show you my config for the powerplug (tun printer on and off when finished):

Moonraker:
```
[power Ender Rechts Power]
type: hue
#   The type of device.  Can be either gpio, klipper_device, rf,
#   tplink_smartplug, tasmota, shelly, homeseer, homeassistant, loxonev1,
#   smartthings, mqtt or hue.
#   This parameter must be provided.
off_when_shutdown: True
#   If set to True the device will be powered off when Klipper enters
#   the "shutdown" state.  This option applies to all device types.
#   The default is False.
off_when_shutdown_delay: 300
#   If "off_when_shutdown" is set, this option specifies the amount of time
#   (in seconds) to wait before turning the device off. Default is 0 seconds.
on_when_job_queued: False
#   If set to True the device will power on if a job is queued while the
#   device is off.  This allows for an automated "upload, power on, and
#   print" approach directly from the slicer, see the configuration example
#   below for details. The default is False.
locked_while_printing: True
#   If True, locks the device so that the power cannot be changed while the
#   printer is printing. This is useful to avert an accidental shutdown to
#   the printer's power.  The default is False.
restart_klipper_when_powered: True
#   If set to True, Moonraker will schedule a "FIRMWARE_RESTART" to command
#   after the device has been powered on. If it isn't possible to immediately
#   schedule a firmware restart (ie: Klippy is disconnected), the restart
#   will be postponed until Klippy reconnects and reports that startup is
#   complete.  Prior to scheduling the restart command the power device will
#   always check Klippy's state.  If Klippy reports that it is "ready", the
#   FIRMWARE_RESTART will be aborted as unnecessary.
#   The default is False.
restart_delay:1
#   If "restart_klipper_when_powered" is set, this option specifies the amount
#   of time (in seconds) to delay the restart.  Default is 1 second.

address:192.168.178.40
#   A valid ip address or hostname of the Philips Hue Bridge. This
#   parameter must be provided.
user:euxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
#   The api key used for request authorization.  This option accepts
#   Jinja2 Templates, see the [secrets] section for details.
#   An explanation how to get the api key can be found here:
#   https://developers.meethue.com/develop/get-started-2/#so-lets-get-started
device_id:29
#   The device id of the light/socket you want to control.
#   An explanation on how you could get the device id, can be found here:
#   https://developers.meethue.com/develop/get-started-2/#turning-a-light-on-and-off
```

Printer conf:
```
[gcode_macro POWER_OFF_PRINTER]
gcode:
  {action_call_remote_method(
    "set_device_power", device="Ender Rechts Power", state="off"
  )}

[delayed_gcode delayed_printer_off]
initial_duration: 0.
gcode:
  {% if printer.idle_timeout.state == "Idle" %}
    POWER_OFF_PRINTER
  {% endif %}

[idle_timeout]
gcode:
  M84
  TURN_OFF_HEATERS
  UPDATE_DELAYED_GCODE ID=delayed_printer_off DURATION=60
```