Arksine/moonraker #161: Just updated the moonraker repo via the Fluidd interface and now it will no longer connect.

ilyakyznesov, 2021-05-12:
[image]

moonraker config
[server]
host: 0.0.0.0
port: 7125
enable_debug_logging: False
config_path: ~/klipper_config
temperature_store_size: 600
gcode_store_size: 1000

[authorization]
enabled: True
cors_domains:
  *.local
  *.lan
  *://app.fluidd.xyz

trusted_clients:
    10.0.0.0/8
    127.0.0.0/8
    169.254.0.0/16
    172.16.0.0/12
    192.168.0.0/16
    FE80::/10
    ::1/128

[history]

[octoprint_compat]

[update_manager]

[update_manager client fluidd]
type: web
repo: cadriel/fluidd
path: ~/fluidd

Arksine, 2021-05-12:
I would need moonraker.log to be useful.  You also may find that the Klipper discord is a better place for support requests than the Moonraker issue tracker, as the issue could be related to Moonraker, a client, your network, etc.

ilyakyznesov, 2021-05-12:
[moonraker.log](https://github.com/Arksine/moonraker/files/6465632/moonraker.log)

the log where it still worked was not saved, since I tried to reinstall everything from scratch

ilyakyznesov, 2021-05-12:
> I would need moonraker.log to be useful. You also may find that the Klipper discord is a better place for support requests than the Moonraker issue tracker, as the issue could be related to Moonraker, a client, your network, etc.

ilyakyznesov, 2021-05-12:
> I would need moonraker.log to be useful. You also may find that the Klipper discord is a better place for support requests than the Moonraker issue tracker, as the issue could be related to Moonraker, a client, your network, etc.

the log where it still worked was not saved, since I tried to reinstall everything from scratch

ilyakyznesov, 2021-05-12:
[moonraker.log](https://github.com/Arksine/moonraker/files/6467069/moonraker.log)

ilyakyznesov, 2021-05-12:
sorry, this is my first time using

Arksine, 2021-05-12:
This log appears to one taken before you updated?  Regardless, you can try a clean install of moonraker to see if it helps.

SSH into your Pi and issue the following commands:
```
cd ~
sudo service moonraker stop
rm -rf moonraker
git clone https://github.com/Arksine/moonraker.git
./moonraker/scripts/install-moonraker.sh -f -c ~/printer_config/moonraker.conf
```

Taudris, 2021-05-13:
I had the same issue of auto update in Fluidd breaking moonraker. Nothing I did in Fluidd or KAIUH fixed it. Following your reinstall instructions solved the issue.

Here's my [moonraker.log](https://github.com/Arksine/moonraker/files/6471855/moonraker.log) from when I was having the issue.

ilyakyznesov, 2021-05-13:
> I had the same issue of auto update in Fluidd breaking moonraker. Nothing I did in Fluidd or KAIUH fixed it. Following your reinstall instructions solved the issue.
> 
> Here's my [moonraker.log](https://github.com/Arksine/moonraker/files/6471855/moonraker.log) from when I was having the issue.

Thank you.  I'm at work now, but as soon as I'm home I'll try

Arksine, 2021-05-13:
Thanks.  That log shows that a new python dependency didn't install correctly during the update.  Most likely there was some kind of networking error that prevented pip from downloading the required package.

I plan to work on a new method of distribution which will catch these kind of errors before an update is performed.

ilyakyznesov, 2021-05-18:
uh, it worked ... I don’t know why it occurred to me, but I disconnected the arduino cable from the rasberry, waited until everything started, plugged it back in and it ... it worked! very strange, thank you all for your help, I hope this "method" will help someone

closed 2021-05-18 by ilyakyznesov