Arksine/moonraker #164: gpio works weird after update

mironfs, 2021-05-17:
today a updated fluidd+klipper+moonraker and now my relay for printer acts weirdly. when i use button in klipper to turn it on it turns on. if i turn it off nothing happens. if i restart moonraker it turns off as it should. so basically i can turn on printer but i have to turn it off via restart moonraker. it worked normally before update

i tried reinstall moonraker from here: https://github.com/Arksine/moonraker/issues/161 didnt help

[moonraker.log](https://github.com/Arksine/moonraker/files/6494608/moonraker.log)

Arksine, 2021-05-17:
I can't reproduce any problems with gpio on my end, however I have an idea of what might be happening.   A recent patch queries device state before any action is taken.  This is to prevent Moonraker from inadvertently taking action in the event that a device is toggled outside of Moonraker.  Your issue suggests that the gpio library on the Pi 4 is returning the device state as off when it is on.  The underlying implementation might not correctly account for the inverted state of the pin.

I notice that there is kernel update available in your log.  I'd suggest updating the system and trying again.  If necessary I can modify the power module to skip the refresh for GPIO devices since it isn't possible to toggle them outside of Moonraker's control, however ideally the gpio library should return the correct value.

mironfs, 2021-05-17:
i did sudo apt update+upgrade right after i posted here. didnt help

Arksine, 2021-05-17:
Ok,  I'd like to check one thing.  If you could add the following option to the `[server]` section of moonraker.conf:
```
enable_repo_debug: True
```
Restart moonraker, connect fluidd, toggle the switch a few times, then upload the resulting log.  You will probably want to remove that option or set it to False afterward, as when it is left on it can create very large logs.

mironfs, 2021-05-17:
[moonraker.log](https://github.com/Arksine/moonraker/files/6495550/moonraker.log)

Arksine, 2021-05-17:
Ugh, my fault.  I mislabeled the option.  Its:
```
enable_debug_logging: True
```
Sorry about that!

Arksine, 2021-05-17:
As an aside, I have a Pi 4 that I put Moonraker on and an led.  Oddly it behaves as expected.  I presume you have tried to reboot the Pi?

mironfs, 2021-05-17:
no problem, yeah i did several times
[moonraker.log](https://github.com/Arksine/moonraker/files/6496277/moonraker.log)

Arksine, 2021-05-17:
Ok thanks.  Lets try to confirm that the refresh change is the culprit.  SSH into your pi and run the following:
```
cd ~/moonraker
git reset --hard 06bea715a3c9ef56152555cb9a4816a21ac136fb
sudo service moonraker restart
```
That will roll back moonraker to commit before the refresh change was applied.  If it works then I'll push a fix, otherwise we are dealing with some other issue.

mironfs, 2021-05-17:
it works now!

Arksine, 2021-05-17:
Ok, thanks.  I will push a fix shortly

Arksine, 2021-05-17:
Ok, pushed what I hope should be a fix.  You can update via update manager if you wish, or you can simply ssh and git pull 
```
cd ~/moonraker
git pull
sudo service moonraker restart
```
Let me know how it goes.  Thanks.

mironfs, 2021-05-18:
im not at home but i vpned to my printer and updated it via manager and based on behaviour (it connects when i flip switch, disconnects when i flip it again or click panic button)its working. thanks!

closed 2021-05-18 by mironfs