Arksine/moonraker #169: File editor destroys hard and symbolic links

adolson, 2021-05-25:
I'm using current Mainsail and Moonraker and Klipper on Raspian Buster.

If you edit a config file that is hard or symbolic linked, and then save the file, the link is destroyed and a new file is written. This is not expected behaviour in Linux - changes should be saved to the linked file, preserving the link.

Arksine, 2021-05-25:
This is the nature of modifying files over http.   We dont actually modify the file in place, we stream the file to a temp directory then move it to the desired location.  This would indeed destroy links.

Moonraker's inotify integration prevents me from moving the file to the location the link points to.  I sync requests with inotify events, if I don't move the file to the requested location I either won't get an event or I won't get the correct event.

adolson, 2021-05-26:
Okay, thanks for the explanation.

closed 2021-05-26 by adolson