Arksine/moonraker #170: UFP handling and Metadata broken

forlack, 2021-05-25:
Describe the bug:
Moonraker does not pull metadata from gcode or ufp files anymore on version v0.6.0-130-g5f7cff9

Expected behavior:
UFP files would become gcode files with a thumnail attached.  Layer information, slicer estimates are displayed in the file manager and print status.

Screenshots:\

Logfiles:\

Relevant Log section from moonraker:

2021-05-24 17:59:04,256 [file_manager.py:_process_file_event()] - Inotify file create: gcodes, /home/pi/gcode_files, CE3_xyzCalibration_cube.ufp
2021-05-24 17:59:04,263 [file_manager.py:_process_file_event()] - Inotify writable file closed: /home/pi/gcode_files/CE3_xyzCalibration_cube.ufp
2021-05-24 17:59:04,292 [file_manager.py:_process_file_event()] - Inotify file delete: gcodes, /home/pi/gcode_files, CE3_xyzCalibration_cube.ufp
2021-05-24 17:59:04,420 [file_manager.py:_process_file_event()] - Inotify file create: gcodes, /home/pi/gcode_files, CE3_xyzCalibration_cube.ufp
2021-05-24 17:59:04,592 [file_manager.py:_process_file_event()] - Inotify file delete: gcodes, /home/pi/gcode_files, CE3_xyzCalibration_cube.ufp
2021-05-24 17:59:04,660 [shell_command.py:run_with_response()] - /home/pi/moonraker-env/bin/python: Traceback (most recent call last):
File "/home/pi/moonraker/scripts/extract_metadata.py", line 754, in
main(args.path, args.filename, args.ufp)
File "/home/pi/moonraker/scripts/extract_metadata.py", line 716, in main
extract_ufp(ufp, file_path)
File "/home/pi/moonraker/scripts/extract_metadata.py", line 696, in extract_ufp
with zipfile.ZipFile(ufp_path) as zf:
File "/usr/lib/python3.7/zipfile.py", line 1222, in init
self._RealGetContents()
File "/usr/lib/python3.7/zipfile.py", line 1289, in _RealGetContents
raise BadZipFile("File is not a zip file")
zipfile.BadZipFile: File is not a zip file

021-05-24 17:59:04,660 [shell_command.py:_check_proc_success()] - Command (/home/pi/moonraker-env/bin/python /home/pi/moonraker/scripts/extract_metadata.py -p /home/pi/gcode_files -f "CE3_xyzCalibration_cube.gcode" -u "/home/pi/gcode_files/CE3_xyzCalibration_cube.ufp") exited with return code 1
2021-05-24 17:59:04,843 [file_manager.py:_process_file_event()] - Inotify writable file closed: /home/pi/gcode_files/CE3_xyzCalibration_cube.ufp
2021-05-24 17:59:04,844 [ioloop.py:_run_callback()] - Exception in callback functools.partial(<bound method IOLoop._discard_future_result of <tornado.platform.asyncio.AsyncIOMainLoop object at 0x75d392b0>>, <Task finished coro=<INotifyHandler._process_inotify_events() done, defined at /home/pi/moonraker/moonraker/components/file_manager.py:1179> exception=FileNotFoundError(2, 'No such file or directory')>)
Traceback (most recent call last):
File "/home/pi/moonraker-env/lib/python3.7/site-packages/tornado/ioloop.py", line 741, in _run_callback
ret = callback()
File "/home/pi/moonraker-env/lib/python3.7/site-packages/tornado/ioloop.py", line 765, in _discard_future_result
future.result()
File "/home/pi/moonraker/moonraker/components/file_manager.py", line 1186, in _process_inotify_events
await self._process_file_event(evt, node)
File "/home/pi/moonraker/moonraker/components/file_manager.py", line 1279, in _process_file_event
await node.complete_file_write(evt.name)
File "/home/pi/moonraker/moonraker/components/file_manager.py", line 825, in complete_file_write
mevt = self.ihdlr.parse_gcode_metadata(file_path)
File "/home/pi/moonraker/moonraker/components/file_manager.py", line 1123, in parse_gcode_metadata
path_info = self.file_manager.get_path_info(file_path)
File "/home/pi/moonraker/moonraker/components/file_manager.py", line 414, in get_path_info
modified = os.path.getmtime(path)
File "/home/pi/moonraker-env/lib/python3.7/genericpath.py", line 55, in getmtime
return os.stat(filename).st_mtime
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/gcode_files/CE3_xyzCalibration_cube.ufp'
2021-05-24 17:59:05,163 [file_manager.py:_process_metadata_update()] - Error running extract_metadata.py
Traceback (most recent call last):
File "/home/pi/moonraker/moonraker/components/file_manager.py", line 1482, in _process_metadata_update
await self._run_extract_metadata(fname, ufp_path)
File "/home/pi/moonraker/moonraker/components/file_manager.py", line 1517, in _run_extract_metadata
result = await scmd.run_with_response(timeout=timeout)
File "/home/pi/moonraker/moonraker/components/shell_command.py", line 240, in run_with_response
self.return_code, stdout, stderr)
components.shell_command.ShellCommandError: Error running shell command: '['/home/pi/moonraker-env/bin/python', '/home/pi/moonraker/scripts/extract_metadata.py', '-p', '/home/pi/gcode_files', '-f', 'CE3_xyzCalibration_cube.gcode', '-u', '/home/pi/gcode_files/CE3_xyzCalibration_cube.ufp']'
2021-05-24 17:59:05,658 [shell_command.py:_check_proc_success()] - Command (/home/pi/moonraker-env/bin/python /home/pi/moonraker/scripts/extract_metadata.py -p /home/pi/gcode_files -f "CE3_xyzCalibration_cube.gcode") successfully finished

Arksine, 2021-05-25:
Generally it is necessary to attach the full moonraker.log to submit a bug report.  That said, this section of the log indicates that there is an issue with the UFP file itself.   The zip utility isn't seeing it as a valid zipfile.

forlack, 2021-05-25:
Ok let me upload a gcode and ufp file and upload the log file.  Did Cura 4.9 break UFP compatibility with moonraker?

forlack, 2021-05-25:
Here is the full log file.  UFP was created in Cura 4.9.1
[moonraker (1).log](https://github.com/Arksine/moonraker/files/6543013/moonraker.1.log)

forlack, 2021-05-26:
And to be clear.  I uploaded a gcode version first and then the UFP.  I do not have any metadata with the gcode version either.  I use to so its either the new Cura version or the new moonraker version.

Arksine, 2021-05-26:
Thanks.  It looks like you are copying the files to a shared folder?  I don't see upload requests in this log, but I do see where inotify is picking up an external file event.  The strange thing is that the source is copying the file to the folder, deleting it, then copying it again.  Its possible that this is corrupting the file.

My recommendation would be to add `[octoprint_compat]` to `moonraker.conf`, remove the share, and try uploading via Cura's Octoprint plugin.

Arksine, 2021-05-26:
To provide an update, I can reproduce this behavior by having Cura write export a gcode/ufp to a sshfs share.  There is something very weird about the way Cura is writing these files.  Its as if they create an empty file just to make sure they can do it.  They then delete the empty file and create the actual file.  This is causing an error that Moonraker cant recover from.

I think I can do some things on my end to handle it and plan to push a fix shortly.

Arksine, 2021-05-26:
Ok, I pushed a change that should hopefully allow you to continue to use your file share if that is what you prefer to do.  I tested it here and it works.

Note that with inotify you shouldn't need to refresh the file manager, it should be notified when the new file is added.

forlack, 2021-05-26:
Awesome thanks Arksine!  Updating now.  Would you still recommend I look at the octoprint plugin for moonraker?

forlack, 2021-05-26:
So it looks like the metadata is working again.  I'm still having issues with UFP files (I only use them for the thumbnails).  I'm going to try the octoprint plugin. 
[moonraker (3).log](https://github.com/Arksine/moonraker/files/6547918/moonraker.3.log)

forlack, 2021-05-26:
I added the octoprint config and cura uploaded directly to mainsail with the thumbnail.  :)  It did not autostart the print, which I prefer because I like to make sure everything is preheated and ready to go.  So while there is still some issue with my old process of using the share directly I will probably use the octoprint config because it saves a couple of clicks :)

Arksine, 2021-05-26:
Thanks for the update.  Its also possible to configure the plugin to start the print, however if you prefer not doing that then no need to change it.

It looks like the remote file system (or Cura) is spamming inotify events.  Before inotify this was probably ok, as long as Cura finished the upload you could manually refresh and everything worked.   Now I'm getting premature events before writing is complete, so Moonraker is trying to process metadata early.  My suspicion is that Cura is responsible for this, because I can copy a large ufp using the file manager and it works ok.  That said, there may be more I can do to mitigate this.

forlack, 2021-05-26:
Cool thanks for investigating this and pointing me to [octoprint] config.  This allows one click to upload with thumbnail.

viddi87, 2021-06-08:
Same thing happening here, i'm exporting directly to shared folder from PrusaSlicer and i need to restart Moonraker to get metadata and thumbnails.

Arksine, 2021-06-08:
Ok, so took a look at this and it appears that PrusaSlicer is doing something different.  It saves the gcode to a temporary file, then renames it.  This never triggers metadata extraction.  I'll fix this when I get a chance, however for the time being it is possible to enable [octoprint_compat] in `moonraker.conf` and configure prusaslicer to export over http.

viddi87, 2021-06-09:
It working now, after the latest update, Big thank you.

DanielJoyce, 2021-06-11:
Rename is atomic. So on Unix/posix this is the proper way to create a file and have it be resistant to corruption.

Basically write to a temp file. Flush. Rename to final filename. Since you only rename once the file is complete, the chance of power outage or crash causing a corrupt file is a lot lot less.

Arksine, 2021-06-11:
Yes, that is generally how its done, however Cura doesn't do it this way.  The issue at hand with PrusaSlicer wasn't due to file corruption or an issue of renaming itself, it was simply a matter of Moonraker's metadata parser not expecting a file with a non-valid gcode extention to be renamed to a file with a valid gcode extension.

closed 2021-12-22 by Arksine