Arksine/moonraker #174: [FR] Power gpio - switch timeout

emtrax-ltd, 2021-06-03:
Emulate the function of a reset button via gpio that will fallback to it's initial state after a specified delay (timeout in n seconds). For example you can short circuit the reset-pin of a auto-power-off-module (MKS PWC, Bigtreetech Relay) via relay or mosfet connected to the gpio to switch on (restart) the printer after a shutdown.

Arksine, 2021-08-28:
Some devices have a similar "timer" functionality built in to their firmware (tasmota and shelly).  Any attempt to implement this for all devices would need to be careful in that it isn't confusing.

JMO, but it would probably be easier to limit something like this to GPIOs.  Moonraker has full control over the GPIOs it initializes.  Their behavior is predictable and we have full control over when and what toggles them.

Networked switches can be accessed and controlled from other software.  Given that we can't exactly predict how they behave its better for "timers" to be implemented in the firmware.

wkeifenheim, 2021-08-28:
_This reply got longer than I anticipated. I started working on code yesterday so I will be as thorough as I can to (hopefully) minimize discussion latency._

I agree with all your points. After working with the `power` component yesterday, and sleeping on it, I had a couple more thoughts on why targeting only the GPIO type for now would be safer. The risk is probably too high to try to solve at the level of the `PrinterDevice` class in `power.py` ([like I started to do](https://github.com/wkeifenheim/moonraker/commit/45baada570e4d14413f52912521ce1a9219fd3b9)) and as was already stated any timer logic is better implemented at the firmware/hardware level. The solution needs more per-device-type consideration.

- I do not own any of the network device types to test with, and building a test suite with simulated hardware for all types is not something I can undertake right now (that would be cool though).

- A ~0.5 second timeout between on/off toggles should work OK with direct (raspberry pi) hardware, but where network latency and other less deterministic scenarios come into play it could be a mess to try to perform a reset action (toggle twice) from Moonraker to networked power relays.

Thanks for pointing out that tasmota and shelly already have some firmware capabilities that could be leveraged, I was planning to look at tasmota code at some point.

### Power Actions
For the power config, how about allowing config for what `actions` are allowed on the device and including that list with the response of the devices `GET` request? I have been running with a `reset` action name in my efforts so far, though I do not have my heart set on it.

Config like:
```yaml
[power device_name]
type: gpio
#   The type of device.  Can be either gpio, tplink_smartplug, tasmota
#   shelly, homeseer, homeassistant, or loxonev1.
#   This parameter must be provided.
enabled_actions:
  on
  off
  toggle
#   The options available to users of the API for use in the 'action' field
#   when setting device state.
#   Available actions are:
#       on - set the device power to 'on'
#       off - set the device power to 'off'
#       toggle - switches the device state from the current state to opposite
#       reset - switches between initial, opposite, and back to initial state with a delay
#   The reset action is currently only supported for the gpio device type.
#   Default includes : on, off, toggle
reset_delay: 0.5
#   Only applies for the 'reset' action. Value is the number of seconds to delay after
#   first state change.
#   The default is 0.5 seconds.
...
```

Getting devices would then include the actions that are enabled for it:
```json
{
    "devices": [
        {
            "device": "printer_gpio_relay",
            "status": "off",
            "locked_while_printing": true,
            "type": "gpio",
            "enabled_actions": ["reset"]
        },
        {
            "device": "green_led",
            "status": "off",
            "locked_while_printing": false,
            "type": "tplink_smartplug",
            "enabled_actions": ["on", "off", "toggle"]
        }
    ]
}
```
With the list of available actions, the UI then has more information with which to render controls.

### Device State
The current design of the `power` component has good variable naming for devices that are only _on_ or _off_ and a request only has one state transition. For my relay, the current state (as it is) is not very important and only the success of toggling twice matters. This has implications for the state/status fields of the device object and API responses that I can think going one of a few ways:

1. State (status) response will be whatever the original state was if the request was successful. If you don't get an error message, you are good to go.
2. State response from a `reset` request will return an 'OK' value if successful.
3. Device state is changed to represent the state of the _device being powered_ by the power relay device. Or the `PrinterPower` class tracks state. In this scenario the reset, or similar command sent to network device, would change the _printer_ power status after completion. 

    Assuming a power relay device is in the `off` initial state, each `reset` requests would return the **printer state**.
    | Request # | Action | Power Device Ops. | Printer State |
    | :---------------: | :---------------: | --------------- | --------------- |
    | 1 | reset | set `on` -> delay -> set `off` | `on` |
    | 2 | reset | set `on` -> delay -> set `off` | `off` |
  
I lean towards options 1 or 2 from the perspective of minimizing changes to the code. However the third option may make more sense when considering the `off_when_shutdown` and `restart_klipper_when_powered` processes.

AntoszHUN, 2022-07-08:
Hi! Is there any progress on this? I would like to operate my BTT Relay module via the RST pin, so that I reset it from the GPIO16 pin. Unfortunately, at the moment, POWER only functions as a switch in the right-hand menu on Fluidd's web interface. I would like to use it as a virtual push button, so to speak (RESET button). Possible?

closed 2022-07-09 by Arksine