Arksine/moonraker #181: [FR] power: on/off gcode config parameter

alchemyEngine, 2021-06-15:
Hey there,

I always run gcode to park my toolhead in a certain spot and disable all stepper before powering off my main PSU via a `[power]` configured relay. It would be ideal for my own workflow if this could be baked into the configuration as e.g. `gcode_on:` and `gcode_off:` config parameters. This could also function as a safety of sorts e.g. in the case of accidentally leaving the steppers powers when updating Moonraker/Klipper. hitting the button, etc.

Thanks for all your work,
-alch3my

Arksine, 2021-06-18:
Interesting, however I think that this is probably something that is better handled by Klipper using a `gcode_macro`.  Moonraker's configuration probably shouldn't contain gcode, and Klipper has robust template support which makes it ideal for implementing custom actions such as moving to the tool.

alchemyEngine, 2021-06-18:
Indeed, it could be integrated into a macro with an `action_call_remote_method`. Unfortunately this makes the very convenient power toggle buttons in fluidd/mainsail much less convenient for me without a configurable set of attached actions. I'll see what cadriel and meteyou think about making those configurable.

closed 2021-06-21 by alchemyEngine