Arksine/moonraker #183: [Feature Request] Schedule start job

tlsomers, 2021-06-15:
Add the possibility to start a job at a later time/date, rather than immediately. Such a system could be used to start a print first thing in the morning (If you're worried about printing while you sleep), or simpler things such as preheating the bed each morning, or shutting down at 12pm if you forgot and no jobs are running.

Optional: Allow the option to either not run if a job is active (default), run after current job finishes, or pause/cancel current job then run this one

Lecso11, 2022-08-04:
I've really like that feature too!

djrampage, 2023-07-25:
Same feature already implemented into marlin fork "SHUI" - delayed start by countdown or in set time. I frequently used it for same reason - my printer is too loud for homies at night and print jobs can take much time.
So i need easy way to make delayed job start too without cron, gcode helpers, home assisant, etc - just by gui button.

djrampage, 2023-09-16:
I made a workaroud with web api.
Create scheduler in cron with command to execute:
curl -X POST http://127.0.0.1:7125/server/job_queue/start

Put your scheduled job into job queue and it automatically starts by cron.

But job queue must be in "ready" status. After startup it is in "paused" status. You can change status with first started job. For example you can use gcode file with single G28 command, put into queue and start queue manually.

I set command to execute daily in 9am and it works. Every day print started automatically if something exists in job queue.