Arksine/moonraker #186: Lot of disconnection with moonraker.

Talha909, 2021-06-25:
I had attached the Pi zero with the board. Its works fine before the update. Now I get lot of disconnection.  Whats the reason behind this. 
[Uploading moonraker (2).log…]()

Arksine, 2021-06-25:
The link to your log is broken.  Please reattempt to upload it.  Thanks.

Talha909, 2021-06-26:
[moonraker.log](https://github.com/Arksine/moonraker/files/6719619/moonraker.log)
Please check.

Arksine, 2021-06-26:
It looks like the problem is a result of the upstream websocket dependency.  According to the log, the current system is a Pi 3. If you attempted to move the SD card from a Pi Zero to a Pi 3 you will want to flash the SD card with a fresh install of FluiddPi.

If you don't want to flash an image you could try clean install of Moonraker.  SSH into your Pi and enter the following:
```
cd ~
sudo service moonraker stop
rm -rf moonraker
git clone https://github.com/Arksine/moonraker.git
./moonraker/scripts/install-moonraker.sh -r
```

Talha909, 2021-06-26:
HI thanks for the response. I did the change the card. Card is already in the PI zero. I just got the update notification. And I update the system. After update. I got connection error message.  When every i gave command I got that error message. I dont have Pi 3

Arksine, 2021-06-26:
The log you uploaded indicates that the system is a Pi 3B+.  It also indicates a version of Moonraker that isn't up to date with the latest version in the repo.

Either something is very wrong or somehow you uploaded the incorrect log.

Talha909, 2021-06-26:
thanks for the answer. 
[moonraker (1).log](https://github.com/Arksine/moonraker/files/6720481/moonraker.1.log)
Please see this. thanks for quick support.

Arksine, 2021-06-27:
Thanks.  That log does reflect Moonraker running on a Pi Zero, however it doesn't seem to show any issues with disconnects.

As it stands now, I can only guess as to what the issue may be:
1) It could be a local network issue that coincidentally popped up after an update
2) I notice quite a few unsafe shutdowns tracked in the log.  Removing power from the Pi without safely powering down can result in file system corruption.  Roughly six weeks ago Moonraker added support for watching the file system through inotify.  If you upgraded from an installation that didn't include inotify it is possible that file system corruption is now causing a problem.
3) It could be another underlying system issue that previously existed, however now Fluidd will disconnect if it cannot ping Moonraker within a specific timeframe.

My suggestion would be to either run fsck on your sdcard or reflash Raspbian/FluiddPi.  If you still have issues and can reproduce them you can report back and upload the log reflecting the problem.

Arksine, 2021-12-22:
As discussion has concluded on this issue I'm going to close it as resolved.  If you need further help I would suggest checking the Klipper discord.

closed 2021-12-22 by Arksine