Arksine/moonraker #187: Software Update section gone after Moonraker update

Roycinger, 2021-07-02:
Hi,

I just updated moonraker via the Fluidd inbuilt software update solution.
After that, I dont see the software update section anymore in Fluidd, neither for Fluidd, Klipper and OS packages. 
Also, I'm getting an error that the Moonraker has failed plugins:
[image]

I'm suspecting that the syntax of the update manager changed.
Before, I had this in my moonraker.conf:
[image]

According to the github documentation, the syntax is different now (I can be wrong here).

Any help is appreciated.

Arksine, 2021-07-02:
Right, those options were deprecated quite some time ago and were recently removed.  It would now look something like this
```
[update_manger]

[update_manager client fluidd]
type: web
repo: cadrie/fluidd
path: /home/pi/fluidd
```

Roycinger, 2021-07-04:
Thank you, I didn't notice that the old syntax style was deprecated. 

This solved it for me!

closed 2021-07-04 by Roycinger