Arksine/moonraker #188: Feature request - allow non official sources or debug mode in update_manager.conf

cooked, 2021-07-02:
currently it is not possible to update klipper, moonraker or any other package that is not from an official repo
modifying the update_manager.conf will render the moonraker "dirty" on next update because there's no "persist_file" option
it is also not possible to enable a "static" section for debug

Arksine, 2021-07-02:
Updates on dirty repos aren't possible as we can't reliably determine if a merge will be successful.

The `[update_manager]` section has an `enable_repo_debug` option that you can set to `True` to disable the origin check.  Keep in mind that if you choose to do this you are operating in dev mode, to report any issues you would need to reproduce on pristine code from official sources.

cooked, 2021-07-03:
> Updates on dirty repos aren't possible as we can't reliably determine if a merge will be successful.
>
> The `[update_manager]` section has an `enable_repo_debug` option that you can set to `True` to [...]

@Arksine the `enable_repo_debug: true` has been added to the "static" sections of the update manager, yet moonraker still reports dirty 
[image]

Arksine, 2021-07-03:
It needs to be in the `[update_manager]` section of `moonraker.conf`.

https://moonraker.readthedocs.io/en/latest/configuration/#update_manager

cooked, 2021-07-03:
> It needs to be in the `[update_manager]` section of `moonraker.conf`.
> 
> https://moonraker.readthedocs.io/en/latest/configuration/#update_manager

:facepalm: yep you're right... still moving that to the [update_manager] section of moonraker.conf gives the same result... moonraker repo still reports dirty (saved file and restarted, restarted moonraker and finally refreshed the update manager)

Arksine, 2021-07-03:
The repo must be pristine.  There can be no modifications to any file within it for the update manager to work.

cooked, 2021-07-03:
@Arksine yes, this point is clear, hence the feature request. what options are there if one wants to sync a non official klipper repo  without breakign the update manager (now it has to get the moonraker repo dirty)? Is this planned for the future? 
I can imagine that getting rid of the "static" update_manager.conf and move those sections into the public moonraker.conf with a big fat disclaimer would serve the purpose. Wdyt?

Arksine, 2021-07-03:
There are no plans to support unofficial repos outside of debug mode.  When `enable_repo_debug` is set to `True` there is no need to modify the static assets in `update_manager.conf`, as the check against the official origin is disabled.

Arksine, 2021-07-03:
It looks like you are attempting to add support for a Pi distribution, presumably it includes a fork of Klipper.  I'll explain in more detail how and why the `update_manger` functions the way it does.

The `enable_repo_debug` option allows Klipper, Moonraker, and client developers to use forks for development and testing.  They can direct their testers to pull their forks and the update manager can pull from unofficial repos.  In this mode Moonraker can update repos with detached heads as well.  No repo configuration is necessary for this to function outside of `enable_repo_debug`.  The `origin` option in the `static` configuration is used for validation only, and it is ignored when in debug mode.  Moonraker will use the origin detected in the repo itself to perform the update.

In production, official repos are validated because Moonraker relies on specific functionality within Klipper to function properly.  It isn't possible for me to guarantee that forks will behave correctly, nor could I contribute to forks to fix any issues that may arise.

cooked, 2021-07-03:
@Arksine aha moment here! thanks for the explanation, and yeah that is exactly my use case: due to the long time in between official Klipper releases we've decided to build a system based on our own fork. 
Regarding the topic at hand, I guess the key point, that was not clear to me, is that the origin: is ignored in debug mode, so moonraker can be left unchanged and pristine indeed. 
I will test it later, still feels a bit of a hack but it is the way it is
Feel free to close this issue and thanks again for taking the time to clarify

cheers

closed 2021-07-19 by cooked