Arksine/moonraker #189: MQTT debug?

nicedevil007, 2021-07-03:
Heya,

just noticed that you implemented mqtt support now! First thank you for this awesome addition!

I already set mine up but it doesn't seem to work, I can't get any message in my node-red instance.
Is there any log where I can take a look at to find any error message?

Arksine, 2021-07-03:
Moonraker has a log, its default location is at `/tmp/moonraker.log`.   If necessary you can add `enable_debug_logging: True` to the `[server]` section in order to enable verbose logging.  This will make the log grow large quickly, so I suggest turning it off when your debugging is complete.

Moonraker responds to API requests over the response topic, it won't publish anything without a request.  Requests are done over the API topic in JSON-RPC format, see the [API documentation](https://moonraker.readthedocs.io/en/latest/web_api/#json-rpc-api-overview) for details.  When debug logging is enabled requests and responses, including their payload, are logged.

In the future I plan to leverage the MQTT component to enable higher level functionality in Moonraker.  For example, I plan to add a `mqtt` type to the power module.  There are plans to implement a `notification` module that will push specific events to topics, and I suspect that there will be a `remote_sensor` module that will allow users to expose various details of environmental sensors to Web Clients.

nicedevil007, 2021-07-04:
That describes everything, I was hoping to get messages out of mqtt like I do over subscriptions to the normal api, but that is not implemented right now.

So I will wait until this is done as well in the future! Thank you for clearing this up, maybe you can add this description to the documents as well?

Arksine, 2021-07-04:
Its hinted at in the documentation (see the note in the [subscription api request](https://moonraker.readthedocs.io/en/latest/web_api/#subscribe-to-printer-object-status)), but I can try to clear it up.  I'm not sure MQTT is a good candidate for rapid status updates, the websocket is a better choice here.   I might consider implementing a throttled version that pushes changes to Klipper/Moonraker state at a slower interval.  If I implement this it would likely have its own topic.

The goal for a `notification.py` component will be to  publish high level events over MQTT to topics configured by the user.  The idea here would be to allow communication with other MQTT clients that are subscribed to specific topic, expecting a particular payload.

nicedevil007, 2021-07-04:
That's absolutely fine. I heated about that a raspberry pi would be able to handle 100s or even 1000s of messages over mqtt. I'm not that interested in every 0.01°C change on the nozzle temp.

What I would like are the main things:

- filament runout
- actual printstate (only publish if changed to prior state)
- temp nozzle (only report if 1°C difference to prior temp)
- bed temp (same as nozzle)
- actual print time / estimated time
- file name that is printed right now

That should be all.

TxBillbr, 2021-08-31:
Following this. I heavily use MQTT/HomeAssistant and would find this quit useful.

Arksine, 2021-12-22:
Moonraker now supports publishing Klipper status asynchronously to a MQTT topic.  All of the requested items can be retrieved the printer object subscriptions.  Klipper pushes subscriptions every 250ms, and only pushes an item if its value has changed.  Of the items requested, the only functionality not available is the 1°C difference in in temperatures.  You will need to add your own logic to your client to discard temperature updates you do not desire.

Moonraker also supports MQTT power devices.

closed 2021-12-22 by Arksine

nicedevil007, 2021-12-22:
Ty very much