Arksine/moonraker #191: Update manager KlipperScreen broken since Update

nicedevil007, 2021-07-08:
Heya,

I recently updated to latest Moonraker version and changed the updatemanager to the new documentation.

It did work for fluidd but not for KlipperScreen. What am I doing wrong?

```
[update_manager KlipperScreen]
type: git_repo
path: /home/pi/KlipperScreen
origin: https://github.com/jordanruthe/KlipperScreen.git
env: /home/pi/.KlipperScreen-env/bin/python
requirements: scripts/KlipperScreen-requirements.txt
install_script: scripts/KlipperScreen-install.sh
```

Arksine, 2021-07-08:
I found a typo.  I will have it corrected shortly.

nicedevil007, 2021-07-08:
I was just on the run to upload my log :D haha

Arksine, 2021-07-08:
Fix is up.  You might need to comment out / remove your clients to get the UM to load, update moonraker, then add them back.

nicedevil007, 2021-07-08:
[image]

it is fixed :) thank you!

closed 2021-07-08 by nicedevil007