Arksine/moonraker #195: Moonraker crash after latest update

BomberX, 2021-07-18:
Hi
Since the update 0.7.1-31, moonraker seems to crash.
Please check the uploaded moonraker.log file. Seems that something is missing to have the networks stats working...
[moonraker.log](https://github.com/Arksine/moonraker/files/6836526/moonraker.log)

Arksine, 2021-07-18:
The version of moonraker reported in your log suggests that the source has been modified 
```
v0.7.1-31-g2f95b0f-dirty
```
Its possible that this was caused by a failed git merge during the update or file system corruption.  You should be able to use either mainsail or fluidd to recover.  If not, ssh into the pi and enter the following commands
```
cd ~/moonraker
git reset --hard origin/master
sudo service moonraker restart
```

BomberX, 2021-07-18:
@Arksine : thanks. Fluidd is not accessible as I have the 'no connection to Moonraker' message. Your workaround did not work either. After I restarted the Pi, I could not access back to anything. It seems that I got a system corruption somewhere and that I will have to rebuild it. Thanks anyway. I will close this ticket.

closed 2021-07-18 by BomberX

Arksine, 2021-07-18:
Sorry to hear that, file system corruption is no fun.  When you get everything reinstalled it should work, but if there is an issue feel free to reopen this ticket and we can work it out.