Arksine/moonraker #197: Feature Request: Server Side Post-Processing Script Support

tuncof, 2021-07-27:
Since neither klipper nor downstream (moonraker, fluidd, mainsail, etc) not (and declared has no plan for) supporting backlash compensation natively, as a compromise, it would be great to do it via server side post processing scripts (python) automatically started after a gcode is uploaded to a virtual sd card folder. Therefore a macro framework for this sort of automated scripts may save people from burden of running them on client side (ie. on slicers), especially for windows users which is a royal PITA. We all use SoBs for klipper and python is already out there, right? So why redundancy?
Cheers

tuncof, 2021-08-01:
Any hope whatsoever?

Arksine, 2021-08-01:
I'm not saying "not ever", however I won't have time to look at something like this in the near future.  I'm open to allowing something like this through templates.  That said, when dealing with large gcode files the experience would likely be better on a desktop class PC.  The real issue on SBCs is the rather slow I/O.

tuncof, 2021-08-02:
While you are right on that most old 1st gen. RPIs are somewhat slow workhorses, I have a bunch of SBCs like Odroid C1+ (4 cores) and Allwinner A20 based ones (Banana PI, Cubieboard 2 -- 2 cores each) which are despite aged too, performances were superior to even recent RPIs. So I don't even bother cross compilation and it never took for me more than 10-15 mins to compile natively a decent mainline fresh kernel with necessary modules on it.  Apparently SBCs are and will be getting faster and faster. This aforementioned compiling performance was even comparable to my ancient Core 2 Duo laptop! So you can give it a chance. It shouldn't be harder than spawning a script when a file shows up in the directory. Originally I intended to use inotify but obviously not worked due to race(?) conditions ie. immediately started to print without delay.

tuncof, 2021-08-02:
To dig in further, I did some quick tests about python script performance. The script under test is a software linear advance code which is a heavily cpu-bound modifier and can be obtined here ([https://www.thingiverse.com/thing:4578664/files](url)) and sample of 24MB gcode is from here ([https://github.com/FABtotum/gcode-samples/tree/master/samples/Advanced](url))

On my Banana PI board script finished its job around 12 minutes with Python3 and 8 minutes with PyPy. And Odroid C1+ scores even half of these. So not that bad for 15+ hours of gcode, eh? Considering this scales linearly with gcode size (~quarter minute per MB at most), I think an acceptable delay performance. I know klipper has already linear advance option but I chose this script rather somewhat as a torture-test for benchmark purposes. Most of my scripts are not so heavy modifier. And there may be other options to use good old compiled C binaries instead of interpreted python scripts.
Regards

tuncof, 2021-08-02:
Sorry if I sounded a bit perseverant and gave any headache to you but for my purposes even a very basic support like that would suffice in moonraker.conf :

[post-processing] or [pre-exec] , etc
script1=/path/to/script
script2=/path/to/script
...

These will be in-place modifier scripts supplied with file name and other environment variables and/or command line args and will return control back to moonraker immediately after handling their job. 

By the way I noticed that I forget to very appreciate your work here which saved us from octoprint-like resource greedy piece of software which are the *real* killer of SBCs by now for my experience. Everything is like a blaze with Klipper, Moonraker, KlipperScreen and Fluidd for me till now. 
Regards