Arksine/moonraker #198: Cannot control homeassistant switch through [switch]

chdgriff, 2021-07-28:
I'm trying to set up PSU control of my 3D printer using a homeassistant compatible smart plug. I followed the documentation and setup my config accordingly:
(From configuration.md)
```
[power homeassistant_switch]
type: homeassistant
address: 192.168.1.126
port: 8123
device: switch.1234567890abcdefghij
token: home-assistant-very-long-token
domain: switch
```

I believe my moonraker.conf is configured correctly since I can see the correct state of the switch in fluidd and I don't get any errors in the log mentioning the homeassistant info command from power.py failing.

However, trying to turn on or off the switch results in an error.
From my moonraker.log file:
```
2021-07-27 23:20:03,088 [power.py:_send_homeassistant_command()] - Error sending homeassistant command: on
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/components/power.py", line 731, in _send_homeassistant_command
    url, method="POST", body=json.dumps(body), headers=headers)
tornado.httpclient.HTTPClientError: HTTP 405: Method Not Allowed
2021-07-27 23:20:03,088 [power.py:set_power()] - Error Setting Device Status: 3D Printer to on
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/components/power.py", line 731, in _send_homeassistant_command
    url, method="POST", body=json.dumps(body), headers=headers)
tornado.httpclient.HTTPClientError: HTTP 405: Method Not Allowed

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/components/power.py", line 378, in set_power
    state = await self._send_power_request(state)
  File "/home/pi/moonraker/moonraker/components/power.py", line 747, in _send_power_request
    await self._send_homeassistant_command(state)
  File "/home/pi/moonraker/moonraker/components/power.py", line 739, in _send_homeassistant_command
    raise self.server.error(msg)
utils.ServerError: Error sending homeassistant command: on
2021-07-27 23:20:03,088 [websockets.py:build_error()] - JSON-RPC Request Error: 400
Error Setting Device Status: 3D Printer to on
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/websockets.py", line 224, in execute_method
    result = await method(conn, *args, **kwargs)
  File "/home/pi/moonraker/moonraker/websockets.py", line 329, in func
    ip_addr=ws.ip_addr, user=ws.current_user))
  File "/home/pi/moonraker/moonraker/components/power.py", line 193, in _handle_batch_power_request
    result[name] = await self._process_request(device, req)
  File "/home/pi/moonraker/moonraker/components/power.py", line 220, in _process_request
    await ret
  File "/home/pi/moonraker/moonraker/components/power.py", line 383, in set_power
    raise self.server.error(msg) from None
utils.ServerError: Error Setting Device Status: 3D Printer to on
```

I verified that my homeassitant REST API isn't acting weird and used Postman to send REST commands to successfully switch on, off, and verify the state of the switch for homeassistant. Anyone have homeassitant switches working or have an idea on how to fix this?

Arksine, 2021-07-28:
Your homeassistant is returning a 405 error when Moonraker sends it a "POST" request.  This functionality was user contributed and I do not personally have homeassistant, but there most be some configuration problem on the homeassistant side that is causing it to reject the request.  When using Postman to validate, did you try sending a POST?

chdgriff, 2021-07-28:
Yes I verified that the post and get request are working

Arksine, 2021-07-28:
Unfortunately I can't explain why Homeassistant would return a 405 to the post request unless the `domian` option is not configured correctly in moonraker.  By default it appears that Moonraker will make the following request turn the switch on ```
POST http://192.168.1.126/api/services/switch/turn_on
```
The `device` is sent in the body and the `token` is applied to the Authorization header.  If this matches what you expect and it still doesn't work perhaps someone who uses the Home Assistant functionality can chime in and be of assistance.

jimmyhideous, 2021-11-05:
Had the same issue. I was able to resolve it by addubg `protocol: https` to the config block.  Otherwise it would fail when trying to connect via http.

[power printerswitch]
type: homeassistant
address: [HA ADDRESS]
port: 8123
**`protocol: https`**
device: switch.device_name
token: [ HA Token]
domain: switch

https://github.com/Arksine/moonraker/blob/61fab42281da627bc14af2e50ed2e026272e5385/moonraker/components/power.py#L385

Arksine, 2021-12-22:
As discussion has concluded on this issue I am going to close it as resolved.  I believe that the `protocol` option is now documented which should help future users.

closed 2021-12-22 by Arksine