Arksine/moonraker #199: OctoPrint Job API Compatibility

Kragrathea, 2021-07-28:
I have a GCode visualization plugin for OctoPrint called  [PrettyGCode](https://github.com/Kragrathea/OctoPrint-PrettyGCode) that I am turning into a stand alone web app. The plan is that app will support both OctoPrint and Moonraker equally. Ideally I would like to code to just one API and Moonrakers Octoprint compatible API would be perfect. Unfortunately /api/job returns a mostly nulled out structure:
{"job": {"file": {"name": null}, "estimatedPrintTime": null, "filament": {"length": null}, "user": null}, "progress": {"completion": null, "filepos": null, "printTime": null, "printTimeLeft": null, "printTimeOrigin": null}, "state": "Operational"}
  
By looking at the code it appears to just be unimplemented. Is there a reason it hasn't been finished? I haven't done much with Python but I could probably finish it using the other parts of the API that do work as a guide.

Arksine, 2021-07-28:
The octoprint compat module is not intended to provide full API compatibility, its purpose is to provide temporary bridge for slicers to be able to upload gcode files straight to Moonraker.  When (if) they get stable native support for Moonraker it is my intention to deprecation that module.

Kragrathea, 2021-07-28:
I see. That leaves me with a bit of a quandry. I had sort of counted on that API as a bridge to getting my viewer working with klipper. And I am going to have to have separate code paths for the OctoPrint and Klipper/Moonraker versions. 

Would you consider letting me add just a few more working parameters at least until you deprecate the module? I already made the few minor changes in my branch. Basically I all I need is the filename and current file position when printing.

https://github.com/Kragrathea/moonraker

Arksine, 2021-07-28:
Sure, I would be ok with that.

Kragrathea, 2021-07-28:
Sweet. That will save me a bunch of time short term. I'll make sure it is finished and then create a pull request.

Thanks Arksine!

Kragrathea, 2021-07-28:
I plan to use the Moonraker APIs eventually. I like the way the push api works. For now the problem is legacy supporting OctoPrint.

Kragrathea, 2021-08-07:
After looking at it for a while I have decided to code directly to the Moonraker push API. I had trouble getting smooth playback with pull APIs. Thanks anyway. :-)

closed 2021-08-07 by Kragrathea