Arksine/moonraker #201: Problem displaying gcode thumbnail generated from Simplifi3D

doktorcb, 2021-07-29:
I was looking for where the problem with the thumbnail is and why they don't appear, after a long time I discovered this.
There is no option to search for thumbnails in the "/home/pi/moonraker/moonraker/components/file_manager/metadata.py" configuration file for the S3D slicer. I copied this part from the Prusaslicer and copied it to the Simplifi3D part and everything works as it should. Please integrate this item permanently, there are enough fans who will appreciate this support.

doktorcb, 2021-08-13:
Ano ten mám nainstalován , ale i přesto když odešlu gcode po krájení do serveru tak mi to minituru nevygeneruje. musím povolit generování miniatur v metadata.py. Ale pokud to udělám, tak server reaguje že kód není čistý. 
Ukázka nastavení scripts:
" C:\3d-thumbs\S3D_Thumbs.exe f=[output_filepath] x=400 y=65 w=953 lw=300 lh=200 sw=32 sh=32
curl -k -H "X-Api-Key: xxxxxxxxxxxxxxxxxxxxxxxxxxxx" -F "select=false" -F "print=true" -F "file=@[output_filepath]" "http://10.0.1.46/api/files/local" {STRIP "; postProcessing"} "

th33xitus, 2021-08-14:
Sorry if my previous comment was unclear.
> Yes, I have it installed, but even so, if I send gcode to the server after slicing, it won't generate a miniature for me.

This is the expected behaviour. Since S3D does not output a thumbnail in its gcode files by default, there was no method implemented yet which allows to parse that thumbnails. This is no problem with moonraker. It's a functionality you expect to be there which does not exist as you already found out.

Arksine, 2021-08-14:
To be clear, there are plugins for Simplify3D that support PrusaSlicer type thumbnails?  Are these plugins officially supported?  I don't mind adding support for thumbnails in S3D if they are officially supported by the slicer.  If they aren't then I don't think its a good idea.

th33xitus, 2021-08-14:
@Arksine In my first post i linked both sources i found during a little bit of research about thumbnail generation for Simplify3D.
The first one is the original script. The second one seems to be a windows port of the first one.

It's the first time i hear about thumbnails in combination with S3D gcode too. I used that slicer for a few years myself and there are no official plugins. Simplify3D doesn't even have a plugin interface or something similar. 

So both those scripts are inofficial 3rd party software which, if i understood correctly, just take a screenshot of the gcode preview, then encode that with base64 and write that to the gcode.

Arksine, 2021-08-14:
Thanks.  IMO it would be better to have native thumbnail functionality from the S3D developers and wait until they have shipped it before implementing anything in Moonraker.  This way there is no need to change/update/deprecate code in Moonraker should S3D choose a different method.

This is how we handled it with Cura.

th33xitus, 2021-08-14:
Maybe there is nothing to worry about that something will ever change in S3D. At least in the current version. Development is stagnating there and people didn't receive an update for that slicer for like 2 years or so. Development in general seems pretty much dead in general for that software. So im pretty sure a native thumbnail functionality won't happen at all. I even doubt there will be a next Simplify3d version.