Arksine/moonraker #203: [Feature Request] Belkin Wemo Smart Plug Support

Dreamwebb, 2021-07-31:
Hi is there any possibility to add Wemo Smart Plug support like the other supported plugs?

Here are the octoprint equivalents

https://plugins.octoprint.org/plugins/wemoswitch/
https://github.com/jneilliii/OctoPrint-WemoSwitch

Vansmak, 2022-01-21:
Do you by chance use homeassistant? You can add any switch from HA.

jimmysdiy, 2023-05-24:
any update on this? I switched from Marlin to Klipper, and I am using Mainsail. I'd love to get my plug to work.

NotAFood, 2023-06-04:
https://gist.github.com/NotAFood/6d864166c7a239ca0f35fb0cd4a2a3bc

Not an official fix but I was able to write a Jinja2 template to emit XML SOAP to interact with a Wemo. May not work if Wemo (has / will have) removed local API access

jimmysdiy, 2023-06-04:
@NotAFood - I just tested and it worked! Thank you so much. I did have to change the IP Address & Port #. My Port was 49152. I the was tested on Wemo Plug v3.