Arksine/moonraker #206: Unable to load klipper, Moonraker Log showing JSON error after Update in Fluiid

NordSee21, 2021-08-08:
Issue:  Updated in app.fluidd.xyz (attached to local RPI Zero W).  Updated RPI OS, Moonraker, and Klipper.  Rebooted since there were RPI OS updates.  Not getting a JSON error in MoonRaker and Klipper will not load.  Attached are the Klipper and Moonraker log files.
[moonraker.log](https://github.com/Arksine/moonraker/files/6951449/moonraker.log)
[klippy.log](https://github.com/Arksine/moonraker/files/6951451/klippy.log)

Arksine, 2021-08-09:
You are experiencing some kind of issue with Klipper starting, thus Moonraker is not able to connect to register one of the endpoints fluidd is attempting to access.

I can see in your Klipper log file that it attempted to build the c_helper, my guess is that it did not build correctly.  This can take a long time on a Pi Zero, if its interrupted this will likely corrupt the file.  My suggestion would be to delete it and restart klipper.  SSH into the pi and enter the following: ```
sudo service klipper stop
rm ~/klipper/klippy/chelper/c_helper.so
sudo service klipper restart
```
Make sure you give it plenty of time to build c_helper.so before shutting down or attempting to restart Klipper again.

KoeRt888, 2021-09-20:
I have the same problem. It started after I upgraded my OS to debian bullseye which included a lot of package bumps.
I'm trying to find out if any package is missing that would have been installed as a dependency.

Arksine, 2021-09-20:
The OP's problem is not an issue with Moonraker.  If you are experiencing a similar issue then the best course of action would be to pursue support on the Klipper discord or discourse.

KoeRt888, 2021-09-20:
Yeah, I'm a hundred percent with you. I thought it was clear from my statement that moonraker works fine and that the resulting JSON_RPC error is a symptom of klipper not starting correctly. Since klipper doesn't even parse the .cfg-file.

KoeRt888, 2021-09-21:
@NordSee21 I fixed it for me by uninstalling and reinstalling cffi in the klippy-env.
`~/klippy-env/bin/pip uninstall cffi
~/klippy-env/bin/pip install cffi`

I know it is not the version that is in the klipper-requirement.txt, but it works for me, ymmv. If you want to reinstall the same version that klipper uses natively append the install command with: `==1.12.2`

Arksine, 2021-12-22:
Hopefully the OP was able to resolve his issue through one of the suggestions above.  As there has been no further discussion I am going to close this issue as resolved.

closed 2021-12-22 by Arksine