Arksine/moonraker #214: After latest update no more timelapse

SA7BNT, 2021-08-22:
After the latest update (v0.7.1-36-gf1523ef) i cant use this snippet anymore in the moonraker.conf
 Something i missed??

[image]

FrYakaTKoP, 2021-08-22:
@SA7BNT  checked the timelapse.py still works with v0.7.1-36 totally fine 

can you post your moonraker.log?

SA7BNT, 2021-08-22:
> @SA7BNT checked the timelapse.py still works with v0.7.1-36 totally fine
> 
> can you post your moonraker.log?

[moonraker.log](https://github.com/Arksine/moonraker/files/7027925/moonraker.log)

FrYakaTKoP, 2021-08-22:
```
  File "/home/pi/moonraker/moonraker/components/timelapse.py", line 7
    <!DOCTYPE html>
    ^
```
It looks like you updated the timelapse script too and downloaded it manually on github. But you need to download the raw script, not the html representation on github

connect to ssh and execute following commands:
```
cd ~/moonraker/moonraker/components/
curl -O https://raw.githubusercontent.com/FrYakaTKoP/moonraker/dev-timelapse/moonraker/components/timelapse.py
```
that should fix the timelapse issue

(also there are alot of klippy disconnects in your log but that's another topic)

SA7BNT, 2021-08-22:
> ```
>   File "/home/pi/moonraker/moonraker/components/timelapse.py", line 7
>     <!DOCTYPE html>
>     ^
> ```
>
> It looks like you updated the timelapse script too and downloaded it manually on github. But [...]

Thanks for help, you make my day. 
Ya i know, but still pretty rookie @klipper 
Will have a closer look. 
Thanks for the hint

closed 2021-08-22 by SA7BNT

FrYakaTKoP, 2021-08-22:
your welcome!

again join the klipper discord there is a lot of information to get and also people willing to help

SA7BNT, 2021-08-22:
> your welcome!
> 
> again join the klipper discord there is a lot of information to get and also people willing to help

thanks mate
Will look after klipper discord